openrazer/openrazer #933: I reinstalled solus and openrazer dos not work annymore.

ItsLeffdePeff, 2019-10-22:
i use solus and i reinstalled it. I had an isue before on solus but that was fixed but now it dos the isue again output from sudo modprobe razerkbd is modprobe: ERROR: could not insert 'razerkbd': Exec format error

z3ntu, 2019-10-22:
Check the `dmesg` for anything from `razerkbd`, that will reveal more. I think this message is normally caused by Secure Boot being enabled.

ItsLeffdePeff, 2019-10-22:
i found this in dmesg [  196.416924] razerkbd: disagrees about version of symbol module_layout

ItsLeffdePeff, 2019-10-22:
I also checked if secure boot was enabled and it was disabled

z3ntu, 2019-10-22:
I guess the Solus maintainers have to rebuild the kernel modules, but I can't help much there. I think the best option is to contact them through whatever support medium they have (IRC, bug tracker?)

closed 2020-02-21 by z3ntu