openrazer/openrazer #934: Support for Razer Viper (wired)

Guyver1wales, 2019-10-22:
### Device Info

* Name: Razer Viper
* Product Number: RZ01-02550100
* Model Number: RZ01-0255 Gaming Mouse

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 007 Device 002: ID 1532:0078 Razer USA, Ltd Razer Viper
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0078 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Viper
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

```
Please post the output of
    ls -lah /dev/input/by-id/
```
[root@fedora30 Downloads]# ls -lah /dev/input/by-id/
total 0
drwxr-xr-x. 2 root root 260 Oct 22 13:18 .
drwxr-xr-x. 4 root root 500 Oct 22 13:18 ..
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-1e7d_KAVE_XTD_Headset-event-if03 -> ../event6
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-Ideazon_Zboard_USB_Gaming_Device-event-if01 -> ../event3
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-Ideazon_Zboard_USB_Gaming_Device-event-kbd -> ../event2
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-Razer_Razer_Viper-event-if01 -> ../event9
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-Razer_Razer_Viper-event-mouse -> ../event7
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-Razer_Razer_Viper-if01-event-kbd -> ../event8
lrwxrwxrwx. 1 root root  10 Oct 22 13:18 usb-Razer_Razer_Viper-if02-event-kbd -> ../event12
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-Razer_Razer_Viper-mouse -> ../mouse1
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-ROCCAT_ROCCAT_Isku_FX-event-kbd -> ../event4
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-ROCCAT_ROCCAT_Isku_FX-if01-event-mouse -> ../event5
lrwxrwxrwx. 1 root root   9 Oct 22 13:18 usb-ROCCAT_ROCCAT_Isku_FX-if01-mouse -> ../mouse0
[root@fedora30 Downloads]# 

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
don't have time currently sorry.

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
haven't used this mouse on windows sorry.

Sondeluz, 2020-01-04:
Hello! I see that this has been quite abandoned...

I recently acquired a wired Razer Viper and I am willing to help. I am attaching here screenshots from Synapse's config tool and a .pcapng which I hope is helpful. I captured the mouse as requested on the Wiki, on a Virtualbox win10 VM (with the usb transfered to it of course) right from a system startup to Synaptics picking it up, I then modified the brightness, changed the color to a Static purple and modified the value third DPI stage, finally I decreased the DPI stages and then increased them back to 5 as before.

I guess this capture may not be completely helpful so please let me know what do you need, I will be happy to help. I may be found on the matrix channel under the same username, I will say hi later!

[razer viper 4th jan screenshots and pcapng.zip](https://github.com/openrazer/openrazer/files/4021976/razer.viper.4th.jan.screenshots.and.pcapng.zip)

Sondeluz, 2020-01-12:
Separate captures with their actions explained, as discussed.
[separate pcapngs.zip](https://github.com/openrazer/openrazer/files/4050805/separate.pcapngs.zip)

raulfunkie, 2020-04-14:
Any updates on the support for the Razer Viper? the app recognizes my other razer mice but not the Viper yet.

I'm actually running this software on a raspberry pi (it's the only linux-like system I have available). Is there anything I can do to help?

z3ntu, 2020-04-19:
PR is here: #1103

commit 9af66a66 (2020-04-24): Add support for Razer Viper

Fixes #934
PR #1103 (2020-04-24): Add support for Razer Viper
Fixes #934
closed 2020-04-24 by z3ntu