openrazer/openrazer #935: Support for Razer Blade Stealth 13 late 2019 GTX

bgm13, 2019-10-23:
### Device Info

* Name: Razer Blade Stealth 13 late 2019
* Product Number: RZ09-03101E72
* Model Number: RZ09-0310

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 003 Device 003: ID 1532:024a Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x024a 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
Please post the output of
    ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 180 Oct 22 19:01 .
drwxr-xr-x 4 root root 600 Oct 22 19:01 ..
lrwxrwxrwx 1 root root  10 Oct 22 19:01 usb-Azurewave_Integrated_Camera_200901010001-event-if00 -> ../event11
lrwxrwxrwx 1 root root  10 Oct 22 19:01 usb-Azurewave_Integrated_Camera_200901010001-event-if02 -> ../event12
lrwxrwxrwx 1 root root   9 Oct 22 19:01 usb-Razer_Razer_Blade-event-if01 -> ../event9
lrwxrwxrwx 1 root root   9 Oct 22 19:01 usb-Razer_Razer_Blade-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Oct 22 19:01 usb-Razer_Razer_Blade-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root  10 Oct 22 19:01 usb-Razer_Razer_Blade-if02-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 Oct 22 19:01 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

NephariusAT-Source, 2019-11-17:
Got mine yesterday. Would be glad to see it working :)

soren121, 2019-11-27:
I got the Mercury White model today with Iris Plus graphics only. The log output for this model is the same as OP's.

echapman2022, 2019-12-04:
I also just got a blade stealth late 2019 and decided to write the support myself. It works on mine, so I created a pull request, but in the mean time, you can find it here: https://github.com/echapman2022/openrazer

To compile it and install, clone the repo, cd into it, type `make`, then `./scripts/build_debs.sh` and the debs will be in `./dist`

doctorblinch, 2019-12-08:
@echapman2022 
Hi! I made steps that you told and installed all .deb, then reinstalled razer polychromatic. But it still says that device is not recognized. I have gtx full HD model.
Thanks =)

echapman2022, 2019-12-08:
The updated code is under the branch feature_stealthlate2019, make sure you clone that one.

NephariusAT-Source, 2019-12-09:
> 
> 
> @echapman2022
> Hi! I made steps that you told and installed all .deb, then reinstalled razer polychromatic. But it still says that device is not recognized. I have gtx full HD model.
> Thanks =)

In case you have any problems:

git clone https://github.com/echapman2022/openrazer.git -b feature_stealthlate2019  

I have the GTX FHD Model and it works smoothly :)

doctorblinch, 2019-12-10:
Yes, thanks @echapman2022 and @NephariusAT-Source, I forgot about the branch))
Everything works fine now.

nhr, 2019-12-12:
I'm very interested in seeing this PR get merged. I'd also be happ y to act as a tester. I am running Fedora and so I'll need help building RPMs from the branch.

What is the process for getting this formally reviewed and merged?

dylangerdaly, 2020-01-03:
I'm on the Razer Blade Sealth Mercury Late 2019, can confirm this works perfectly.

```
bash-5.0# pwd
/sys/devices/pci0000:00/0000:00:06.0/usb2/2-8/2-8:1.2/0003:1532:024A.0004
bash-5.0# echo -n -e "\xFF\xFF\xFF" > matrix_effect_static
bash-5.0# 
```

commit 29307dc9 (2020-01-10): Add support for Razer Blade Stealth Late 2019

Fixes #935
closed 2020-01-10 by z3ntu