openrazer/openrazer #936: openrazer can't find devices on new Arch Kernel

Plarpoon, 2019-10-24:
[https://i.imgur.com/RxVXBbD.png](url)

openrazer's daemon is not able to see any of the razer devices connected to the machine, i've followed all of the methods like re-plugging the devices once the system was booted and being sure that my user had the right groups.

I'm quite sure this is a kernel problem since a new version just came out and it's having issues with it, mind checking it out please?

lah7, 2019-10-24:
I'm using an Arch derivative (Manjaro) and haven't had any problems with `5.3.7-2-MANJARO`.

Are there any errors in `dmesg`? Are there any errors when you probe the driver?

    sudo modprobe razerkbd

Plarpoon, 2019-10-28:
sorry for the long delay, tomorrow i should be able to get my hands on the desktop and i will let you have the results

Plarpoon, 2019-10-28:
dmesg doesn't show any error, these are the results i have on the razer products:
`[    2.734732] hid-generic 0003:1532:0227.0004: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Huntsman] on usb-0000:0a:00.0-2/input0
[    2.734734] input: Razer Razer Mamba Elite System Control as /devices/pci0000:00/0000:00:1c.4/0000:09:00.0/usb5/5-2/5-2:1.1/0003:1532:006C.0002/input/input6
[    2.734780] input: Razer Razer Mamba Elite as /devices/pci0000:00/0000:00:1c.4/0000:09:00.0/usb5/5-2/5-2:1.1/0003:1532:006C.0002/input/input7
[    2.746965] hid-generic 0003:1532:006C.0002: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Mamba Elite] on usb-0000:09:00.0-2/input1
[    2.751762] input: Razer Razer Mamba Elite as /devices/pci0000:00/0000:00:1c.4/0000:09:00.0/usb5/5-2/5-2:1.2/0003:1532:006C.0003/input/input9
[    2.756668] input: Razer Razer Huntsman Keyboard as /devices/pci0000:00/0000:00:1c.6/0000:0a:00.0/usb7/7-2/7-2:1.1/0003:1532:0227.0005/input/input10
[    2.777862] raid6: avx2x4   gen() 38417 MB/s
[    2.811387] hid-generic 0003:1532:006C.0003: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Mamba Elite] on usb-0000:09:00.0-2/input2
[    2.817924] usb 1-6.2: new full-speed USB device number 4 using xhci_hcd
[    2.834519] raid6: avx2x4   xor() 22037 MB/s
[    2.868005] input: Razer Razer Huntsman Consumer Control as /devices/pci0000:00/0000:00:1c.6/0000:0a:00.0/usb7/7-2/7-2:1.1/0003:1532:0227.0005/input/input11
[    2.873134] input: Razer Razer Huntsman System Control as /devices/pci0000:00/0000:00:1c.6/0000:0a:00.0/usb7/7-2/7-2:1.1/0003:1532:0227.0005/input/input12
[    2.878362] input: Razer Razer Huntsman as /devices/pci0000:00/0000:00:1c.6/0000:0a:00.0/usb7/7-2/7-2:1.1/0003:1532:0227.0005/input/input13
[    2.882017] hid-generic 0003:1532:0227.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Huntsman] on usb-0000:0a:00.0-2/input1
[    2.891188] raid6: avx2x2   gen() 34342 MB/s
[    2.893773] input: Razer Razer Huntsman as /devices/pci0000:00/0000:00:1c.6/0000:0a:00.0/usb7/7-2/7-2:1.2/0003:1532:0227.0006/input/input14
[    2.896531] hid-generic 0003:1532:0227.0006: input,hidraw5: USB HID v1.11 Mouse [Razer Razer Huntsman] on usb-0000:0a:00.0-2/input2
`
While modprobe doesn't print anything on screen

z3ntu, 2019-10-29:
What about `lsmod | grep razer` ?

Plarpoon, 2019-10-29:
still no results

z3ntu, 2019-10-30:
If modprobe succeds, then the module should show in lsmod.
You can always try to rebuild the dkms modules, maybe something has broken there: 
`sudo dkms remove openrazer-driver/2.6.0 --all && sudo dkms install openrazer-driver/2.6.0`

Plarpoon, 2019-10-30:
> If modprobe succeds, then the module should show in lsmod.
> You can always try to rebuild the dkms modules, maybe something has broken there:
> `sudo dkms remove openrazer-driver/2.6.0 --all && sudo dkms install openrazer-driver/2.6.0`

just tested and still no results. I do appreciate all the help i'm receiving tho, thank you a lot!

z3ntu, 2019-10-30:
If modprobe succeeds and lsmod doesn't show the module that has just been loaded (and there aren't any errors in the dmesg regarding that), then there's something seriously wrong where I can't help you unfortunately...

Plarpoon, 2019-11-02:
Gonna wait that this gets elevated, maybe someone else can help. Still, thank you a lot for everything you have done up to now z3ntu

leualemax, 2019-11-03:
same problem here: 
```
Your kernel headers for kernel 5.3.7-2-MANJARO cannot be found at
/usr/lib/modules/5.3.7-2-MANJARO/build or /usr/lib/modules/5.3.7-2-MANJARO/source.
```

leualemax, 2019-11-03:
`sudo pacman -S linux53-headers` and a reboot solve the problem.

closed 2020-04-01 by stale[bot]