openrazer/openrazer #937: DKMS module build fails after gcc update

parkerlreed, 2019-10-24:
### Distribution

Arch Linux

### Support

- [*] Have you tried rebooting?
- [*] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

gcc 9.2.0-3
gcc-libs 9.2.0-3

```
DKMS make.log for openrazer-driver-2.6.0 for kernel 5.3.6-arch1-1-fsync (x86_64)
Thu 24 Oct 2019 02:06:01 PM EDT

:: Compiling OpenRazer kernel modules
========================================
make -C /lib/modules/5.3.6-arch1-1-fsync/build M=/var/lib/dkms/openrazer-driver/2.6.0/build/driver modules
make[1]: Entering directory '/usr/lib/modules/5.3.6-arch1-1-fsync/build'
  CC [M]  /var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.o
cc1: error: incompatible gcc/plugin versions
cc1: error: fail to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.o] Error 1
make[1]: *** [Makefile:1626: _module_/var/lib/dkms/openrazer-driver/2.6.0/build/driver] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.3.6-arch1-1-fsync/build'
make: *** [Makefile:25: driver] Error 2
```

parkerlreed, 2019-10-24:
Just realized all my DKMS are failing to install. Issue lays elsewhere.

closed 2019-10-24 by parkerlreed

z3ntu, 2019-10-24:
I don't have the 9.2.0-3 version installed yet, as it's only in testing. Additionally you have a custom kernel installed, so maybe try rebuilding that (so the headers get rebuilt)? But yes, as you've realized, this isn't a problem with openrazer :)