openrazer/openrazer #940: Support for Razer Huntsman

piedude, 2019-10-27:
### Device Info

* Name: Razer Huntsman
* Product Number: RZ03-02520200-R3U1
* Model Number: RZ03-0252

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 005: ID 1532:0227 Razer USA, Ltd Razer Huntsman
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0227 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Huntsman
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 240 Oct 26 20:04 .
drwxr-xr-x 4 root root 640 Oct 26 20:04 ..
lrwxrwxrwx 1 root root  10 Oct 26 20:04 usb-Razer_Razer_Huntsman-event-if01 -> ../event26
lrwxrwxrwx 1 root root  10 Oct 26 20:04 usb-Razer_Razer_Huntsman-event-kbd -> ../event22
lrwxrwxrwx 1 root root  10 Oct 26 20:04 usb-Razer_Razer_Huntsman-if01-event-kbd -> ../event23
lrwxrwxrwx 1 root root  10 Oct 26 20:04 usb-Razer_Razer_Huntsman-if02-event-mouse -> ../event27
lrwxrwxrwx 1 root root   9 Oct 26 20:04 usb-Razer_Razer_Huntsman-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Oct 26 20:04 usb-Razer_Razer_Mamba_Tournament_Edition-event-if01 -> ../event18
lrwxrwxrwx 1 root root  10 Oct 26 20:04 usb-Razer_Razer_Mamba_Tournament_Edition-event-mouse -> ../event16
lrwxrwxrwx 1 root root  10 Oct 26 20:04 usb-Razer_Razer_Mamba_Tournament_Edition-if01-event-kbd -> ../event17
lrwxrwxrwx 1 root root  10 Oct 26 20:04 usb-Razer_Razer_Mamba_Tournament_Edition-if02-event-kbd -> ../event18
lrwxrwxrwx 1 root root   9 Oct 26 20:04 usb-Razer_Razer_Mamba_Tournament_Edition-mouse -> ../mouse1

### Packet Captures
Will update when I get these.

### Synapse Screenshots
Will update when I get these.

nightsky30, 2019-12-14:
See #792

z3ntu, 2019-12-15:
Duplicaate of #739 , pull request for adding support is at #792

closed 2019-12-15 by z3ntu