openrazer/openrazer #945: Razer Naga Trinity 1532:0067 not loaded in 2.60 on Ubuntu

LeSyl21, 2019-11-06:
Hi,

I'have install via PPA on ubuntu Bionic "18.04.3 LTS"

lsusb | grep 1532

The daemon log show was NAGA is loaded:
2019-11-06 14:06:02 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2019-11-06 14:06:02 | razer                          | DEBUG    | Loaded device specification: RazerNagaChroma ------------------------- (1532:0053)
2019-11-06 14:06:02 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2019-11-06 14:06:02 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2019-11-06 14:06:02 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)

I can't see my mouse even itr's list as supported: https://github.com/openrazer/openrazer#device-support

Thx in advance

z3ntu, 2019-11-06:
Hi, support for the Razer Naga Trinity was only added after v2.6.0 has been released, that's why it's not included yet. It will be included in the next release but in the meantime you can use our daily PPA, see https://openrazer.github.io/#ubuntu

LeSyl21, 2019-11-06:
Thx a lot :-D done and work

z3ntu, 2019-11-06:
Great!

closed 2019-11-06 by z3ntu