openrazer/openrazer #946: Mouse pad Razer Firefly V2

dottantgal, 2019-11-08:
### Device Info

* Name:  Razer Firefly V2
* Product Number: RZ02-03020100
* Model Number: RZ02-0302

### Device Information
```
Bus 003 Device 006: ID 1532:0c04 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0c04 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               2.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1

```

```
lrwxrwxrwx 1 root root   9 nov  8 10:54 usb-RAZER_Razer_Firefly_V2-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 nov  8 10:54 usb-RAZER_Razer_Firefly_V2-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 nov  8 10:54 usb-RAZER_Razer_Firefly_V2-mouse -> ../mouse1

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

CobyBE, 2020-04-17:
Hello!
I would like to contribute to this project, I have this mousepad and was able to save some PCAPs (static colour change, wave effect).
What is the best way to provide these? Also the tutorial part is missing how to write the driver or else I'd like to try my hand on it.

I was mainly interested in getting the wave effect to work under Linux and found this ticket.

z3ntu, 2020-04-19:
@CobyBE You can just upload the pcaps directly to github (it doesn't like the .pcapng extension but packing them in a .zip should work).

For the implementation, you can follow other device support commits, e.g. 8ec0f649d6e5081d70cce58e6eeb64f3a85605da (you can ignore the `scripts/driver/firefly_hyperflux/test.py` file there completely)

If you have any questions, I'm happy to help.

CobyBE, 2020-04-20:
Thanks for the example! I'll create a branch and try to follow the hyperflux support branch implementation

CobyBE, 2020-04-21:
When I have modified the driver classes, how do I deploy them? Just use the build_deps.sh script and then sudo dpkg the *deb files?
Or do I need to use that Makefile in the root folder?

z3ntu, 2020-04-24:
@CobyBE No you don't have to use the Makefile. Yes you run the build_debs.sh script and use `sudo dpkg -i *.deb` to install them. Then it's best to reboot to fully "activate" the changes.

If you want a quicker workflow, e.g. when you're editing the driver, you can just run `make` in the root of the directory which builds a bunch of `.ko` files in the `driver/` directory. You can then use e.g. `sudo rmmod razerkbd && sudo insmod ./driver/razerkbd.ko` to load the freshly built kernel module. Then it's still advisable to unplug your device and plug it back in (otherwise the udev rules won't fix up permissions as needed by the daemon)

CobyBE, 2020-04-29:
Thanks for the help, I managed to get it working and have created a pull request :)
https://github.com/openrazer/openrazer/pull/1123

lravelo, 2020-05-12:
hello.  I just purchased one of these and am running on MX Linux (based on Debian).  I'd love to contribute to getting this device supported if possible (not a developer by any stretch but can defend myself)

CobyBE, 2020-05-14:
@lravelo I have created a branch to add support for this mousemat, there still seems to be a small error with restarting the daemon after coming back from sleep/relogin but the rest works. Bug can be worked around by unplugging and restarting the daemon, then replugging the mousemat;

https://github.com/openrazer/openrazer/pull/1123

Be advised that this is still in development

lravelo, 2020-05-14:
@CobyBE yeah I've been looking at that and been trying to figure out how to use your branch.  I've installed openrazer via the debian package.  I'm guessing that I have to fully remove that and install directly from github?  not sure how to go about doing that.

lravelo, 2020-05-14:
@CobyBE so I removed everything and reinstalled from your fork of the project and checked out this branch and it looks like it's working well.  I can see this now in polychromatic and razergenie.

CobyBE, 2020-05-15:
@lravelo that's great news! :) Enjoy your mousemat :)

lravelo, 2020-05-16:
@CobyBE might've spoke too soon.  I got myself a blackwidow keyboard today and as soon as I rebooted after plugging it in, I no longer see the mat in polychromatic or razergenie.  Tried removing everything and started from scratch but no dice :-(

CobyBE, 2020-05-17:
@lravelo As a workaround, unplug the mousmat, restart daemon and then replug the driver. I'm currently looking into why this happens and how to fix it

lravelo, 2020-05-17:
@CobyBE tried that. It did not work.  Will uninstall all of the GUI's and openrazer itself, reboot, reinstall everything without plugging in the mousemat and let's see if it works.