openrazer/openrazer #947: Devices not recognized after recent update

0tcmct0, 2019-11-13:
Hello,

After a recent update to openrazer ( using 2.6.0-1), My devices are no longer recognized. I'm running Manjaro linux with kernel 5.2.21-1.

Output of  $sudo dkms install openrazer-driver/2.6.0:

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
KERNELDIR=/lib/modules/5.2.21-1-MANJARO/build make driver...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.2.21-1-MANJARO (x86_64)
Consult /var/lib/dkms/openrazer-driver/2.6.0/build/make.log for more information.

Output of make.log:

make -C /lib/modules/5.2.21-1-MANJARO/build M=/var/lib/dkms/openrazer-driver/2.6.0/build/driver modules
make[1]: Entering directory '/usr/lib/modules/5.2.21-1-MANJARO/build'
  CC [M]  /var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.o
cc1: error: incompatible gcc/plugin versions
cc1: error: fail to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.o] Error 1
make[1]: *** [Makefile:1597: _module_/var/lib/dkms/openrazer-driver/2.6.0/build/driver] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.2.21-1-MANJARO/build'
make: *** [Makefile:25: driver] Error 2

So it seems something broke with the new version of openrazer with gcc? Not completely sure. Any help would be much appreciated! Thank you for you time.

0tcmct0, 2019-11-13:
Fixed by updating to kernel version 5.3.8-3.

z3ntu, 2019-11-13:
Closing then.

closed 2019-11-13 by z3ntu