openrazer/openrazer #949: Support for Razer Kraken Kitty Edition

qaqaqaqaowh, 2019-11-16:
### Device info
- Name: Razer Kraken Kitty Edition

### Command outputs

```
Bus 001 Device 014: ID 1532:0521 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0521 
  bcdDevice            0.70
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken Kitty Edition
  iSerial                 3 00000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0123
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0072
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID            10
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               6 Sidetone
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                11
        bSourceID              10
[... 235 lines cut ...]
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     214
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0025  1x 37 bytes
        bInterval               1
```
```
lrwxrwxrwx 1 root root   9 Nov 16 12:21 usb-Razer_Razer_Kraken_Kitty_Edition_00000000-event-if03 -> ../event3
```

999eagle, 2019-12-21:
Hi,
I've captured USB packets for the Razer Kraken Kitty Edition, I hope these help. If you need more captures, I can provide them.

[pcaps.zip](https://github.com/openrazer/openrazer/files/3991387/pcaps.zip)

z3ntu, 2019-12-26:
Thanks, the captures are fine.

Any idea what the device with the USB ID 1532:0f19 is? That seems to have been connected during the capture as well.

999eagle, 2019-12-26:
On my PC, `lsusb -d 1532: -v` gives me two devices, the Kraken Kitty itself and the Kraken Kitty Chroma Control. Listing devices by id gives this output:
```
lrwxrwxrwx root root   9 B Thu Dec 26 11:47:03 2019  usb-Razer_Razer_Kraken_Kitty_Chroma_Control-event-if00 ⇒ ../event7
lrwxrwxrwx root root   9 B Thu Dec 26 11:47:05 2019  usb-Razer_Razer_Kraken_Kitty_Edition_00000000-event-if03 ⇒ ../event9
```

I've attached the full lsusb output here. The headset is the only Razer device I have connected and both devices are gone when I unplug it.
[kitty_lsusb.txt](https://github.com/openrazer/openrazer/files/4002302/kitty_lsusb.txt)

Sirmentio, 2020-05-10:
Hey, I have a Razer Kraken Kitty Edition myself, is there any way I can contribute?

Sirmentio, 2020-05-10:
Documentation said that a screenshot of all available options are good, I hope these help.
[image]
[image]
[image]
[image]
[image]
[image]

I've omitted the mixer, because for the most part it's just audio volume, and I don't think there'll be support for THX spatial surround any time soon on Linux.

Mikeprotein, 2020-05-11:
Good afternoon, I'm relatively new to this, is there any support for the Razer Kraken Kitty on Linux yet? If so, how can I get it?

Kind regards :-)

Sirmentio, 2020-05-15:
Not quite yet, I'm not entirely sure how the process of this works but I'd assume someone would need to make a pull request with the driver code using the information me and others have given. That might take a bit and my heart goes out to someone who would do this for one of the more obscure devices.

keleuk, 2020-05-23:
I'm starting to write the driver for this device, but am at a bit of an impasse. Likely due to the 4 independent LEDs, the protocol for this one is much more similar to the matrix-based keyboard protocol than the one for the other Krakens, and not particularly compatible with the current kraken driver framework. I'm not sure if we'd rather I try and fit the behavior into the existing driver, create a new driver specifically for the kitty kraken (and other possible future matrix headphones), or do something else that I'm not thinking of right now.

depesz, 2020-06-18:
Any idea on progress? I just got these for my wife, and it would be cool to be able to change colors :)

GentlemanSch, 2020-07-11:
I'm also interested in this, for the same reason as depesz . How can we get this device supported?

999eagle, 2020-09-21:
@z3ntu I'm currently trying my hand at a PR for this. The Kraken Kitty Edition seems to do everything using Chroma though, so I'd have to special case every single function in `razerkraken_driver.c`. Would it be better to special case everything there or put it into a new driver file like `razerkrakenchroma_driver.c`?

tbaumann, 2020-10-02:
Thanks a lot @999eagle this is very exciting.
I will try to test that. If it makes sense.

Can I nudge you to add support for the **Razer Base Station Chroma** too? It's such a good pairing with the headset and I got it as a set. :blush:

tbaumann, 2020-10-02:
```
tilmanb@zuse:~/git/openrazer$ python3 ./examples/list_devices.py
Found 1 Razer devices

Razer Kraken Kitty Edition:
   type: headset
   serial: HN1943D12008311
   firmware version: v1.0
   driver version: 2.8.0
```
Looking good. Getting excited.

tbaumann, 2020-10-02:
```
tilmanb@zuse:~/git/openrazer$ python3 ./examples/advanced_effect.py 
Found 1 Razer devices

Setting Razer Kraken Kitty Edition to effect breath_random
```
Works

closed 2020-10-11 by z3ntu

Kusoneko, 2021-06-02:
Hello,

I would like to ask if any of you probed the device for support of the other Chroma effects besides None, Static and Breath (Single, Dual & Random) when adding support for the device, as I'm currently trying to figure out if it's possible to add the other effects or if I'm wasting my time trying to add support for them to the driver of this device.

Thank you,
Kusoneko

999eagle, 2021-06-03:
Hi @Kusoneko,
I've tried a few of the other effects while implementing support for the Kraken Kitty Edition, but at least for me none of the other effects worked if I recall correctly. It's been a while though since I've worked on this, so maybe there is a way to get them working without software based lighting. I can look into it again some time this weekend.

Kusoneko, 2021-06-04:
Hi @999eagle.

I've opened the following issues in openrazer and polychromatic on the subject, listing the effects Razer advertises the Kitty Edition to have. I had no reply on openrazer, but on polychromatic, there's one contributor trying to help me figure out how to try to add effects to the device in openrazer to attempt to figure out which are software effects and which are hardware effects. I would of course welcome your help in this as I am in no way an expert on writing drivers or on the openrazer drivers at all and I do happen to be working a lot this month, so finding free time to tinker in the driver isn't easy.

https://github.com/polychromatic/polychromatic/issues/328
https://github.com/openrazer/openrazer/issues/1520

commit 2b248fcd (2020-10-11): Add support for Razer Kraken Kitty Edition

Fixes #949