openrazer/openrazer #951: No device detected on Razer Blade Stealth 2019 GTX 1650 model

doctorblinch, 2019-11-21:
### No device detected on Razer Blade Stealth Late 2019 GTX 1650 model.
*OS:* Pop OS 19.10 (Ubuntu based OS)
Daemon version 2.6.0.
I've already tried to reinstall needed packages.
I've tried both RazerGenie and Polychromatic.

z3ntu, 2019-11-24:
The output of `lsusb | grep 1532` please

doctorblinch, 2019-11-24:
@z3ntu 

**$ lsusb** 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 1532:024a Razer USA, Ltd Razer Blade
Bus 003 Device 002: ID 13d3:56d5 IMC Networks Integrated Camera
Bus 003 Device 004: ID 8087:0026 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
**$ lsusb | grep 1532**
Bus 003 Device 003: ID 1532:024a Razer USA, Ltd Razer Blade

z3ntu, 2019-11-24:
Duplicate of #935 then

closed 2019-11-24 by z3ntu