openrazer/openrazer #952: Cannot install branch feature.nagaepicchroma.0x003e

navin2012, 2019-11-23:
Hi, I am currently using Pop_OS 19.10 and am trying to use my Naga Epic Chroma mouse. When I initially downloaded OpenRazer it did not come with support for my mouse. I found a branch (feature.nagaepicchroma.0x003e) that solves my issue. Problem is that I am new to github and linux so I am unsure how to get this installed. I've tried copying the files from this branch and pasting onto my existing OpenRazer folder but to no avail. The log files still does not list the naga epic chroma.

I'm sure this has to do with my copy and pasting, if so, what is the proper way to installing this branch?

Thank you!

z3ntu, 2019-11-23:
You can follow https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint for that.

navin2012, 2019-11-24:
Thank you for the advice. I have tried that and it worked like a charm. However, Polychromatic still says that the device is not recognised. I Ctrl-F the log files (attached) and still do not see the Naga Epic Chroma as a loaded device. But reading the drivers loaded on my OpenRazer folder I can see that my mouse has been accounted for. Any thoughts? 
[razer.log](https://github.com/openrazer/openrazer/files/3883096/razer.log)

z3ntu, 2019-11-24:
Did you restart the daemon (or reboot) after installing the new packages?

navin2012, 2019-11-24:
Yes to both. Is it possible that the older OpenRazer files are still persisting? Is there a way to start fresh? Delete (or uninstall) OpenRazer and re-do it? Thank you.

z3ntu, 2019-11-24:
If you've only installed openrazer using dpkg/apt, you can just remove the packages with dpkg/apt. If you've installed it using some other method (some make install target), then you have to manually remove the files that are spread around everywhere in the file system.

closed 2020-04-23 by stale[bot]