openrazer/openrazer #953: Polychromatic applications not opening (Arch, gnome, Blade 15)

SVaeth, 2019-11-23:
### Distribution

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I used the following commands: 
`pacaur -S openrazer-meta`
restarted the laptop
`pacaur -S polychromatic`

and cannot open the controller application.

Some relevant outputs:
```
sudo dkms openrazer-driver/2.6.0x86_64
Module openrazer-driver/2.6.0 already installed on kernel 5.3.12-arch1-1/x86_64
```
```
lsusb | grep 1532:
Bus 001 Device 005: ID 1532:0245 Razer USA, Ltd Razer Blade
```

```
razer-service -F
bash: razer-service: command not found
```
According to the site, my device is the 2019 Blade Mercury (could have sworn it was the Advanced but eh) and is listed as being supported. Is there something I missed? If you need any further outputs, logfiles let me know. I would like to get up and running asap, as my backlights have shut off on my keyboard and have not come back on since the laptop fell asleep earlier

SVaeth, 2019-11-23:
Something to further help with finding the issue, the output of running the polychromatic-controller directly:

```
/usr/bin/polychromatic-controller 
Traceback (most recent call last):
  File "/usr/bin/polychromatic-controller", line 26, in <module>
    import distro
ModuleNotFoundError: No module named 'distro'

```

z3ntu, 2019-11-24:
Install `python-distro`, I forgot to add that dependency to the AUR package.

z3ntu, 2019-11-24:
Fixed (or just update the polychromatic package on your system).

closed 2019-11-24 by z3ntu

SVaeth, 2019-11-24:
Wait, it's not working yet. I'm getting this issue now, and the tray-applet won't even open and just hangs if I run the binary from the terminal
 
[image]
[image]

z3ntu, 2019-11-24:
Does `sudo modprobe razerkbd` succeed? What does `openrazer-daemon -Fv` output?

SVaeth, 2019-11-24:
Not sure what it did, but `sudo modprobe razerkbd` seems to have succeeded, at least no input

```
openrazer-daemon -Fv
Unable to lock on the pidfile.
```

So the daemon is running

SVaeth, 2019-11-24:
I found this in the logs
```
2019-11-24 03:22:49 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0245.0005/device_type, file is not owned by plugdev
```

adding log file. My user is part of plugdev and I have rebooted since. Ignore that first message
[razer.log](https://github.com/openrazer/openrazer/files/3883346/razer.log)

z3ntu, 2019-11-24:
That's issue #877 then

SVaeth, 2019-11-24:
Are you sure it's related to the issue polychromatic is having starting applications/recognizing the drivers?