openrazer/openrazer #954: Linux Mint / Razer Core X Chroma

Lor3nzoMartinez, 2019-11-27:
I am attempting to use the openrazer project with my Razer Core X Chroma and I have gotten it to work on fedora but I ould really enjoy it on Mint. When I look under the device manager it is only showing the MX 150 on-board gpu (I have a razer blade stealth late 2019 13") but on fedora I was able to authorize the thunderbolt port but that doesn't seem to be the case on Mint is there a way to allow communication on the thunderbolt port through he command line or some other way?

z3ntu, 2019-12-15:
No idea, sorry. I don't own a thunderbolt device myself so I've never used it. You should be able to find help in the forum of the distro of your liking probably.

closed 2019-12-15 by z3ntu

turbos-and-tacos, 2019-12-15:
What is your TB security profile in BIOS? You might want to disable security on TB devices to get a baseline if you are able to see it at all. 

You can use `boltctl` to see what is recognized by your machine. I'm on Ubuntu, but for me it looks like:
```
$ boltctl 
 ● Razer Core X Chroma
   ├─ type:          peripheral
   ├─ name:          Core X Chroma
   ├─ vendor:        Razer
   ├─ uuid:          d9010000-0090-8d08-23dd-b41c94851823
   ├─ status:        authorized
   │  ├─ domain:     domain0
   │  └─ authflags:  none
   ├─ authorized:    Sat 14 Dec 2019 11:55:09 PM UTC
   ├─ connected:     Sat 14 Dec 2019 11:55:09 PM UTC
   └─ stored:        no

 ● Razer Core X Chroma #2
   ├─ type:          peripheral
   ├─ name:          Core X Chroma
   ├─ vendor:        Razer
   ├─ uuid:          d4030000-0070-6d18-23dd-b41c18541123
   ├─ status:        authorized
   │  ├─ domain:     domain0
   │  └─ authflags:  none
   ├─ authorized:    Sat 14 Dec 2019 11:55:10 PM UTC
   ├─ connected:     Sat 14 Dec 2019 11:55:10 PM UTC
   └─ stored:        no
```

`boltctl` Should let you authorize it if you are able to see it.