openrazer/openrazer #955: Polychromatic Tray Applet not opening in Gnome  3.34.1

SVaeth, 2019-11-27:
### Distribution
Arch

Gnome 3.34.1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I run `/usr/bin/polychromatic-tray-applet` in command line, but tray-applet, and it runs but does not open. I did some digging and found these issues that may be related:
https://github.com/polychromatic/polychromatic/issues/128
https://github.com/syncthing/syncthing-gtk/issues/534 

If Gnome no longer supports App indicators, does this mean it no longer supports Polychromatic Tray Applet? Is there anything I can do to work around this?

lah7, 2019-11-27:
My understanding is that AppIndicators moved to a GNOME Shell extension.

Try installing the package `gnome-shell-extension-appindicator` and enable/add it to GNOME.

SVaeth, 2019-11-28:
So I pulled that from the aur earlier, but I could not find anywhere to enable it in GNOME Tweaks, and my theory is because they removed support for app indicators

z3ntu, 2019-12-15:
Closing this as it's not an openrazer issue, but a [polychromatic issue](https://github.com/polychromatic/polychromatic/).

closed 2019-12-15 by z3ntu