openrazer/openrazer #956: Input lag

Fl0ux, 2019-11-27:
### Distribution

Ubuntu 19.10

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Device : Razer Naga Trinity

When pressing multiple keys on my keyboard (Roccat Iksu FX) at same time and pressing multiple mouse buttons (numpad), it freezes my system for the time keys are pressed. Xorg process goes up to 60/80% of the CPU.

I don't know how to provide you more information, please let me know.

parkerlreed, 2019-11-28:
openrazer should only be controlling the LEDs. Does it do teh same without the openrazer driver loaded?

Fl0ux, 2019-11-28:
My bad, it does not come from openrazer. Any idea of what I should do to investigate ?

z3ntu, 2019-12-01:
No idea, sorry. I'm closing this issue because it isn't an issue with this driver :)

closed 2019-12-01 by z3ntu