openrazer/openrazer #957: How to read the battery status

Algram, 2019-11-28:
### Distribution
Fedora 30

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I think this is the best place to ask my question and to get proper support:

I have a Razer Mamba Wireless (1532:0072). Now I know that this device is currently not supported. What I would like to ask is, to get a point on how I can request the `battery level` from the mouse.

This would allow me to write a simple script to notify me when the mouse needs to be charged.

Can you give me some hints on how to get started on that issue?

Thanks!

z3ntu, 2019-12-01:
Hi, unfortunately this isn't easily possible, as you need to send USB frames to the device and receive them again to get e.g. the battery level (which is what the driver is doing). Of course you can just patch the openrazer driver to do this for you, but official support won't be included until all the hardware is working - and as that is a newer Razer device, the packet captures from Razer Synapse aren't useful anymore for writing other drivers as the 'commands' we use aren't used by Synapse anymore and as such don't know which ones are used. See #747

Alternatively (which would probably be easier), you can try https://github.com/z3ntu/razer_test which is 'next-gen openrazer' basically ^^ but that project doesn't have support for the battery info yet, but I can add it if needed. I don't have a wireless Razer device myself so that's why I didn't add it yet. With that "razer_test" project, you can then use a simple shell or python script to query the different values from the mouse via the D-Bus interface.

Algram, 2019-12-01:
@z3ntu Thanks for the detailed explanation. Once you added support for the Mamba Wireless and the Battery Info I would gladly try out the razer_test project. A python script that returns the battery status is exactly what I need!

closed 2020-03-30 by Algram