openrazer/openrazer #958: Support for Razer Huntsman TE

shofwan, 2019-11-28:
### Device Info

* Name: Huntsman Tournament edition
* Product Number: RZ03-03080100
* Model Number: RZ03-0308

### Device Information
```
    lsusb -d 1532: -v
Bus 001 Device 002: ID 1532:0243 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd                                                                                                                                                                                                                                     
  idProduct          0x0243                                                                                                                                                                                                                                                    
  bcdDevice            2.00                                                                                                                                                                                                                                                    
  iManufacturer           1                                                                                                                                                                                                                                                    
  iProduct                2                                                                                                                                                                                                                                                    
  iSerial                 3                                                                                                                                                                                                                                                    
  bNumConfigurations      1                                                                                                                                                                                                                                                    
  Configuration Descriptor:                                                                                                                                                                                                                                                    
    bLength                 9                                                                                                                                                                                                                                                  
    bDescriptorType         2                                                                                                                                                                                                                                                  
    wTotalLength          109                                                                                                                                                                                                                                                  
    bNumInterfaces          4                                                                                                                                                                                                                                                  
    bConfigurationValue     1                                                                                                                                                                                                                                                  
    iConfiguration          0                                                                                                                                                                                                                                                  
    bmAttributes         0xa0                                                                                                                                                                                                                                                  
      (Bus Powered)                                                                                                                                                                                                                                                            
      Remote Wakeup                                                                                                                                                                                                                                                            
    MaxPower              500mA                                                                                                                                                                                                                                                
    Interface Descriptor:                                                                                                                                                                                                                                                      
      bLength                 9                                                                                                                                                                                                                                                
      bDescriptorType         4                                                                                                                                                                                                                                                
      bInterfaceNumber        0                                                                                                                                                                                                                                                
      bAlternateSetting       0                                                                                                                                                                                                                                                
      bNumEndpoints           1                                                                                                                                                                                                                                                
      bInterfaceClass         3 Human Interface Device                                                                                                                                                                                                                         
      bInterfaceSubClass      1 Boot Interface Subclass                                                                                                                                                                                                                        
      bInterfaceProtocol      1 Keyboard                                                                                                                                                                                                                                       
      iInterface              0                                                                                                                                                                                                                                                
        HID Device Descriptor:                                                                                                                                                                                                                                                 
[... cut ...]
lrwxrwxrwx 1 root root   9 Nov 29 00:07 usb-Razer_Razer_Huntsman_Tournament_Edition_00000000001A-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Nov 29 00:07 usb-Razer_Razer_Huntsman_Tournament_Edition_00000000001A-if02-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 Nov 29 00:07 usb-Razer_Razer_Huntsman_Tournament_Edition_00000000001A-if02-mouse -> ../mouse1
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

inputbim, 2020-01-16:
can confirm device not recolonised

z3ntu, 2020-05-02:
Fixed with #1010

closed 2020-05-02 by z3ntu