openrazer/openrazer #959: Windows 10 Support

LeithalWeapon, 2019-12-01:
### Distribution

### Support
Blackwidow Ultimate 2016 Edition dropped my mouse in a cup of tea and accidentally messed up some plastic such a n00b. Okay so seriously it makes me cringe having this razer software on my computer potentially keylogging everything into some database in China or Russia or even worse the Americans... Okay so now that I'm done that silly rant I'm going to get down on my hands and knees and beg you to make this wonderful software work for windows I guarantee you that people like myself will love you if you and your awesome coding minions do this. PLEASE AND THANK YOU also have a good end of the year if you can.

- [ ] Have you tried rebooting? Doesn't get rid of windows or the razer software... Joke
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? 

 Lots of Pain From Razer Software Sorry For Not Following The Format.

Danhyal-Masood, 2019-12-07:
unlikely since it would require a complete rewrite .

ZjemCiKolege, 2019-12-10:
I add something about deathadder essential on windows no option for 125hz but in master guaide already explain 3 options for 125,500,1000 thats joke

z3ntu, 2019-12-13:
Hi, I have a rewrite of OpenRazer at https://github.com/z3ntu/razer_test/ which also compiles (and works!) on Windows. It's currently really rough to install (see https://github.com/z3ntu/razer_test/wiki/Windows) but it generally works. As I don't normally use Windows, I won't spend too much time on making the installation procedure nicer, but I would really love some contributions on that front!

Wthebinder, 2019-12-27:
@z3ntu I really appreciate all you have done for this community and all of the time you have put into this project. I would really like to use Razer Test for my Windows PC but I don't understand how to install it. If you have a minute, I was wondering if you could help me out as I am a completely new to this. Many thanks and happy holidays!

z3ntu, 2019-12-29:
So theoretically (according to the instructions I wrote some time ago):
* Go to https://ci.appveyor.com/project/z3ntu/razer-test-pnxsa , choose the latest `arch=x64, compiler=msvc2017` build, click `Artifacts`, download and extract the zip.
* Do the same for https://ci.appveyor.com/project/z3ntu/razergenie
* Download the dbus stuff from https://private.z3ntu.xyz/win-dbus.zip and extract
* Copy the `libdbus-1-3.dll` from win-dbus into the razer_test folder
* Copy the `libdbus-1-3.dll` from win-dbus into the RazerGenie folder
* Start the dbus daemon in a cmd window and in another cmd window, start razer_test.exe: https://github.com/z3ntu/razer_test/wiki/Windows
* Then you should be able to launch razergenie.exe and it *should* work.

I have no experience with Windows services & installers so I likely won't make it much more user friendly in the short future, contributions are very welcome on that front!

Also note, that razer_test currently only supports a small number of devices (there are a couple of open PRs on the razer_test repo for a few more devices but it's still not much)

CalcProgrammer1, 2020-02-19:
I added OpenRazer support to my OpenRGB project.  I'm using a reworked version of an old wrapper to build OpenRazer on Windows:

https://github.com/CalcProgrammer1/openrazer-win32

My OpenRGB program:

https://gitlab.com/CalcProgrammer1/OpenRGB

You will need to use Zadig to install the WinUSB driver on the last interface of all your Razer devices for it to work.  The benefit of this method is that it shares code with OpenRazer upstream, any time a new device is added I can rebuild the DLL and it will work.

closed 2020-07-07 by z3ntu