openrazer/openrazer #964: Kernel 5.4.2 - Ubuntu

davidc502, 2019-12-06:
### Distribution  -- Ubuntu 18.04.03

/var/lib/dkms/openrazer-driver/2.6.0

I purchased a Razor Ornata keyboard and a Razor Basilisk mouse.

### Support

- [YES ] Have you tried rebooting?  Yes
- [YES ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?  Yes

Installed drivers and front end and noticed "Device not recognized".  Immediately went to troubleshooting. The first thing I tried was  "sudo dkms install openrazer-driver/2.6.0"

It looks like Kernel 5.4.2 isn't supported?  What is the latest kernel that is supported?

[code]
davidc502@Ryzen-3900x:~$ sudo dkms install openrazer-driver/2.6.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
KERNELDIR=/lib/modules/5.4.2-050402-generic/build make driver...(bad exit status: 2)
ERROR (dkms apport): kernel package linux-headers-5.4.2-050402-generic is not supported
Error! Bad return status for module build on kernel: 5.4.2-050402-generic (x86_64)
Consult /var/lib/dkms/openrazer-driver/2.6.0/build/make.log for more information.
[/code]

Thank you,

David

SveSop, 2019-12-06:
Weird. I am using kernel 5.4.2 atm, and did not have any issues. I am using Razer Deathadder Elite and no keyboard.
`openrazer-driver, 2.6.0, 5.4.2-cybmod2, x86_64: installed`

What kernel/how did you compile? (Or kernel ppa).

PS. I am also using Ubuntu 18.04

davidc502, 2019-12-06:
I'm using 5.4.2 Mainline, and use the Ubuntu Kernel Update Utility to make those kernel changes.

SveSop, 2019-12-06:
> I'm using 5.4.2 Mainline, and use the Ubuntu Kernel Update Utility to make those kernel changes.

Well, i dunno what would be the difference between my self-hacked/patched kernel and mainline tho, or if it is just something that have to do with the device. As i said, my Razer Deathadder Elite and openrazer 2.6.0 seems to compile and run like usual.

In case you want to compare, my current .config is here: [https://github.com/SveSop/kernel_cybmod_tree/blob/master/.config](https://github.com/SveSop/kernel_cybmod_tree/blob/master/.config)

OBS! This config does NOT work with a AMD processor, and from your text above, it seems as you are running Ryzen, so don't just try to use the .config as-is tho :)

davidc502, 2019-12-06:
This machine is X570 Motherboard (AMD), and there has been several issues since it came out, but is getting better over time.
Thanks for the config, I'll look it over.

z3ntu, 2019-12-13:
I don't really know what's wrong on your machine and why dkms is printing that message, but generally dkms looks for the kernel headers at `/usr/lib/modules/$(uname -r)/build/` so please make sure the headers exist there.

davidc502, 2019-12-13:
Can I use a symbolic link?

davidc502@Ryzen-3900x:~$ ls -l /var/lib/dkms/openrazer-driver/2.6.0/build/driver
total 464
-rw-r--r-- 1 root root    485 Jul 29 09:28 Makefile
-rw-r--r-- 1 root root  39309 Jul 29 09:28 razerchromacommon.c
-rw-r--r-- 1 root root   8760 Jul 29 09:28 razerchromacommon.h
-rw-r--r-- 1 root root   7627 Jul 29 09:28 razercommon.c
-rw-r--r-- 1 root root   4468 Jul 29 09:28 razercommon.h
-rw-r--r-- 1 root root  19749 Jul 29 09:28 razercore_driver.c
-rw-r--r-- 1 root root   1265 Jul 29 09:28 razercore_driver.h
-rw-r--r-- 1 root root 121672 Jul 29 09:28 razerkbd_driver.c
-rw-r--r-- 1 root root   4575 Jul 29 09:28 razerkbd_driver.h
-rw-r--r-- 1 root root  31689 Jul 29 09:28 razerkraken_driver.c
-rw-r--r-- 1 root root   4849 Jul 29 09:28 razerkraken_driver.h
-rw-r--r-- 1 root root 130119 Jul 29 09:28 razermouse_driver.c
-rw-r--r-- 1 root root   2992 Jul 29 09:28 razermouse_driver.h
-rw-r--r-- 1 root root  27704 Jul 29 09:28 razermousemat_driver.c
-rw-r--r-- 1 root root    853 Jul 29 09:28 razermousemat_driver.h
-rw-r--r-- 1 root root  26385 Jul 29 09:28 razermug_driver.c
-rw-r--r-- 1 root root    918 Jul 29 09:28 razermug_driver.h
-rw-r--r-- 1 root root    785 Jul 29 09:28 usb_hid_keys.h

z3ntu, 2019-12-13:
Okay so after reading through the dkms source, it looks like dkms is explicitly checking `apport.packaging.is_distro_package(kernel_package)` - see https://github.com/dell/dkms/blob/master/dkms_apport.py#L48-L60
BUT this seems to be only the "error handling code" which runs when a compilation has failed (apport is the crash reporter from Ubuntu) and as such not relevant.

Can you post the contents of the file `/var/lib/dkms/openrazer-driver/2.6.0/build/make.log` ?

davidc502, 2019-12-14:
Here ya go

[code]
In file included from ./include/linux/bug.h:5:0,
                 from ./include/linux/mmdebug.h:5,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/slab.h:15,
                 from /var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.c:22:
./arch/x86/include/asm/bug.h:35:22: error: expected identifier or ‘(’ before string constant
  asm_inline volatile("1:\t" ins "\n"    \
                      ^
./arch/x86/include/asm/bug.h:79:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, BUGFLAG_WARNING|(flags));  \
  ^~~~~~~~~~
./include/asm-generic/bug.h:98:3: note: in expansion of macro ‘__WARN_FLAGS’
   __WARN_FLAGS(BUGFLAG_ONCE |   \
   ^~~~~~~~~~~~
./include/linux/fs.h:2825:2: note: in expansion of macro ‘WARN_ON_ONCE’
  WARN_ON_ONCE(1);
  ^~~~~~~~~~~~
In file included from <command-line>:0:0:
././include/linux/compiler_types.h:210:24: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/bug.h:35:2: note: in expansion of macro ‘asm_inline’
  asm_inline volatile("1:\t" ins "\n"    \
  ^~~~~~~~~~
./arch/x86/include/asm/bug.h:79:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, BUGFLAG_WARNING|(flags));  \
  ^~~~~~~~~~
./include/asm-generic/bug.h:98:3: note: in expansion of macro ‘__WARN_FLAGS’
   __WARN_FLAGS(BUGFLAG_ONCE |   \
   ^~~~~~~~~~~~
./include/linux/fs.h:2825:2: note: in expansion of macro ‘WARN_ON_ONCE’
  WARN_ON_ONCE(1);
  ^~~~~~~~~~~~
./include/linux/fs.h: In function ‘filemap_nr_thps_dec’:
././include/linux/compiler_types.h:210:24: error: expected ‘(’ before ‘__inline’
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/bug.h:35:2: note: in expansion of macro ‘asm_inline’
  asm_inline volatile("1:\t" ins "\n"    \
  ^~~~~~~~~~
./arch/x86/include/asm/bug.h:79:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, BUGFLAG_WARNING|(flags));  \
  ^~~~~~~~~~
./include/asm-generic/bug.h:98:3: note: in expansion of macro ‘__WARN_FLAGS’
   __WARN_FLAGS(BUGFLAG_ONCE |   \
   ^~~~~~~~~~~~
./include/linux/fs.h:2834:2: note: in expansion of macro ‘WARN_ON_ONCE’
  WARN_ON_ONCE(1);
  ^~~~~~~~~~~~
In file included from ./include/linux/bug.h:5:0,
                 from ./include/linux/mmdebug.h:5,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/slab.h:15,
                 from /var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.c:22:
./arch/x86/include/asm/bug.h:35:22: error: expected identifier or ‘(’ before string constant
  asm_inline volatile("1:\t" ins "\n"    \
                      ^
./arch/x86/include/asm/bug.h:79:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, BUGFLAG_WARNING|(flags));  \
  ^~~~~~~~~~
./include/asm-generic/bug.h:98:3: note: in expansion of macro ‘__WARN_FLAGS’
   __WARN_FLAGS(BUGFLAG_ONCE |   \
   ^~~~~~~~~~~~
./include/linux/fs.h:2834:2: note: in expansion of macro ‘WARN_ON_ONCE’
  WARN_ON_ONCE(1);
  ^~~~~~~~~~~~
In file included from <command-line>:0:0:
[... 169 lines cut ...]
  ^~~~~~~~~~
./include/asm-generic/bug.h:90:19: note: in expansion of macro ‘__WARN_FLAGS’
 #define __WARN()  __WARN_FLAGS(BUGFLAG_TAINT(TAINT_WARN))
                   ^~~~~~~~~~~~
./include/asm-generic/bug.h:115:3: note: in expansion of macro ‘__WARN’
   __WARN();      \
   ^~~~~~
./include/linux/usb.h:1969:3: note: in expansion of macro ‘WARN_ON’
   WARN_ON(usb_pipeout(pipe));
   ^~~~~~~
/var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.c: In function ‘razer_attr_write_matrix_custom_frame’:
/var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.c:1598:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
             report.transaction_id.id = 0x80; // Fall into the 2016/blade/blade2016 to set device id
             ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.c:1599:9: note: here
         default:
         ^~~~~~~
scripts/Makefile.build:265: recipe for target '/var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.o' failed
make[2]: *** [/var/lib/dkms/openrazer-driver/2.6.0/build/driver/razerkbd_driver.o] Error 1
Makefile:1652: recipe for target '/var/lib/dkms/openrazer-driver/2.6.0/build/driver' failed
make[1]: *** [/var/lib/dkms/openrazer-driver/2.6.0/build/driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.3-050403-generic'
Makefile:23: recipe for target 'driver' failed
make: *** [driver] Error 2
[/code]

z3ntu, 2019-12-14:
That looks like an error caused by something weird in the kernel headers, and not by some code in openrazer.. 
Also I'm running kernel 5.4.2 myself and the build works absolutely fine there. So unfortunately I can't help you with this.

closed 2019-12-14 by z3ntu

SveSop, 2019-12-14:
Yep. I second that. No compilation problems with kernel 5.4.2 nor 5.4.3 for me with DKMS either. (Ubuntu 18.04).

@davidc502 I have not tested the mainline kernel, but build my own, so i am not sure if that follows the same directions as ubuntu distro kernel... But afaik the regular distro kernel has 3 things you need to install to get all the kernel headers/modules. 
linux-headers-5.0.0-x, linux-headers-5.0.0-x-generic and linux-modules-5.0.0-x-generic

The "headers" are optional afaik, as the only thing you need is image + modules to get a running system. The headers are "source", and what you and DKMS are looking for. Make sure that you have both: `linux-headers-5.4.3-050403_5.4.3-050403.201912130841_all.deb` and `linux-headers-5.4.3-050403-generic_5.4.3-050403.201912130841_amd64.deb`

Ubuntu "Headers" are in `/usr/src/linux-headers-xxx`, while the modules are in `/lib/modules/xxx`

davidc502, 2019-12-14:
Much thanks... I have several things going on right now, but will circle back around to this at some point.
I do compile source code, but have never done my own kernel, so I may just look at doing that instead of using mainline.

kimocoder, 2019-12-17:
This issue is related to a kernel coming down upstream.

Take a look at some sources below and it all makes sense.. 1 solution is to recompile the kernel, really though solution to go for if you ask me.

https://github.com/iovisor/bcc/commit/2d1497cde1cc9835f759a707b42dea83bee378b8
[Google search finds 'em all](https://www.google.com/search?sxsrf=ACYBGNRYHKcHlNgBBkcXQzNBOeIBDiisFw%3A1576606338770&source=hp&ei=ghr5Xdn0LMaqrgTSk6S4CA&q=.%2F.%2Finclude%2Flinux%2Fcompiler_types.h%3A210%3A24%3A+error%3A+expected+%E2%80%98%28%E2%80%99+before+%E2%80%98__inline%E2%80%99&oq=.%2F.%2Finclude%2Flinux%2Fcompiler_types.h%3A210%3A24%3A+error%3A+expected+%E2%80%98%28%E2%80%99+before+%E2%80%98__inline%E2%80%99&gs_l=psy-ab.3...348.348..574...0.0..0.114.114.0j1......0....2j1..gws-wiz.TDq7e8Japq8&ved=0ahUKEwjZvZWmpL3mAhVGlYsKHdIJCYcQ4dUDCAU&uact=5)

SveSop, 2019-12-17:
> New enough versions of gcc allow one to overrule the estimated size by
using "asm inline" instead of just "asm".

Could this mean that it may be a compiler issue? I mean, when i compile the kernel myself i use `gcc version 9.2.1 20191102 (Ubuntu 9.2.1-17ubuntu1~18.04.1)` but i think the upstream 18.04 kernel might be compiled using gcc-8?

Dunno what "new enough" means in the quote tho, cos it might just aswel be something daft like >gcc 5.4 or some shit :)

kimocoder, 2019-12-17:
That's correct.

qysnn, 2021-10-06:
I ran into the same error. Google search gives https://answers.launchpad.net/ubuntu/+source/v4l2loopback/+question/692404 .It turns out that my `gcc` is outdated. After upgrading `gcc` with `sudo apt-get install gcc g++`, the error is gone.