openrazer/openrazer #965: i cant follow the openSUSE link for fedora

wingzero-KSEC, 2019-12-06:
### Distribution

### Support

- [ Y] Have you tried rebooting?
- [ Y] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

<S0, I'll start off by saying that i am SUPER new to Linux. It's been a challenge and a lot of fun. But i keep running into an issue when trying to get the drivers for my Razer Blackwidow chroma V2 tournament edition. I can get to the sight just fine, but every time I follow the link for the openSUSE buildservice for fedora to follow the instructions I get this "Error connecting to http://api.infra.opensuse.org//search/published/binary/id?match=project='hardware%3Arazer'+and+name='openrazer-meta': Failed to open TCP connection to api.infra.opensuse.org:80 (getaddrinfo: Name or service not known)" and I mean its every time. ive tried switching browsers, and OSs (im dual booting) and its the same thing. 

Any help would be greatly appreciated.>

wingzero-KSEC, 2019-12-06:
I thought I put it in the initial post, but im running fedora 31

z3ntu, 2019-12-06:
Temporary failure on the end of the opensuse build service, it seems like. Try again later please.

closed 2020-04-06 by z3ntu