openrazer/openrazer #966: LanceheadTE left and right lighting

kparrish, 2019-12-08:
### Linux Mint 19.1 "Tessa" (Ubuntu 18.04)

/var/lib/dkms/openrazer-driver/2.6.0
kernel 4.15

### Support

- [Yes ] Have you tried rebooting?
- [Yes ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

In PolyChromatic and RazerGenie scroll and logo lighting work, but left and right lighting are missing.

Digging around the object I found where the logo is getting sourced:
```
In [65]: device.fx.__dict__['misc'].__dict__['_lighting_dbus'].__dict__
Out[65]: 
{'_dbus_interface': 'razer.device.lighting.logo',
```
but couldn't find left or right.
 
Scroll and logo are implemented in nagahexv2 and deathadder_chroma but lanceheadte is the only one with left and right.

Based on:
```bash
 kparrish@raiden  ~/projects/scripts/mouse/openrazer   master 
 19:57:03  grep -rnw './' -e 'setLeftStatic'
./daemon/openrazer_daemon/dbus_services/dbus_methods/lanceheadte.py:110:@endpoint('razer.device.lighting.left', 'setLeftStatic', in_sig='yyy')
./pylib/openrazer/client/devices/__init__.py:112:            'lighting_left_static': self._has_feature('razer.device.lighting.left', 'setLeftStatic'),
 kparrish@raiden  ~/projects/scripts/mouse/openrazer   master 
 19:57:22  grep -rnw './' -e 'setScrollStatic'
./daemon/openrazer_daemon/dbus_services/dbus_methods/nagahexv2.py:177:@endpoint('razer.device.lighting.scroll', 'setScrollStatic', in_sig='yyy')
./daemon/openrazer_daemon/dbus_services/dbus_methods/deathadder_chroma.py:348:@endpoint('razer.device.lighting.scroll', 'setScrollStatic', in_sig='yyy')
./daemon/openrazer_daemon/misc/effect_sync.py:150:                    effect_func = getattr(self._parent, 'setScrollStatic', None)
./pylib/openrazer/client/devices/__init__.py:100:            'lighting_scroll_static': self._has_feature('razer.device.lighting.scroll', 'setScrollStatic'),
```
my best guess is that left and right are missing from the elif cascade in misc/effect_sync.py

lah7, 2019-12-08:
I'm not sure on RazerGenie's status, but Polychromatic v0.3.12 doesn't support `left` or `right` zones. The new version in development does (v1.0.0) - try testing the tray applet in the [dev-1.0.0 branch](https://github.com/polychromatic/polychromatic/tree/dev-1.0.0).

In OpenRazer's Python library, the object is `[device].fx.misc.left` and `[device].fx.misc.right` alongside `scroll_wheel` and `logo`, which can be found when poking in the `python3` interpreter:

```python
import openrazer.client
a = openrazer.client.DeviceManager()
device[0].fx.misc.left.static(0, 255, 0)
```

kparrish, 2019-12-08:
Ah thanks, found left and right were added to MiscLighting class in commit 69f5ab9 after 2.6.0, updating to daily worked.

closed 2019-12-08 by kparrish