openrazer/openrazer #968: Razer Blade Stealth touchpad not working anymore on Ubuntu 16.04

Gnucki, 2019-12-09:
### Distribution

Linux gnuckini 4.15.0-72-generic #81~16.04.1-Ubuntu SMP Tue Nov 26 16:34:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have a Razer Blade Stealth bought on 2017-08-29.
My touchpad doesn't work any more on my Ubuntu 16.04 for about one week.
I don't know the exact day because I don't use it every days, but I'm sure it was working on the 1st of december.
I use a dual boot with windows and it works on that OS so I thought of a driver problem.
Of course, I use your package to the latest stable version (with automatic updates).

Thanks for your help!

lah7, 2019-12-09:
OpenRazer is a driver for LED support, so it shouldn't have anything to do with the touchpad functionality.

Does the problem still happen after uninstalling OpenRazer / rebooting the computer?

Consider checking which packages were upgraded around that time by looking at `/var/log/apt/history.log`, one of those (possibly a kernel or firmware package) is bound to be the problem.

Gnucki, 2019-12-09:
Oh, OK!
Sorry, didn't understand that.
Anyway, thanks for the quick reply and advice!

terrancehanlon, 2020-01-09:
@Gnucki did you by chance find a fix? Mine touchpad also stopped working around the same time as yours. I cant narrow down the package install that caused it.

Gnucki, 2020-01-12:
Hi, @terrancehanlon !
I hadn't time yet to find a solution to my issue.
Someone else asked me for the same problem on Ubuntu 18.04.
The first to find will tell it to the others ! :)

Gnucki, 2020-01-15:
Solved by last kernel update!

closed 2020-01-15 by Gnucki