openrazer/openrazer #974: Support for Razer Nari Ultimate

gboddin, 2019-12-13:
### Device Info

* Name: Razer Nari Ultimate
* Product Number: RZ04-02670100-R3M1
* Model Number: RC30-0267(01)

### Device Information
```
Bus 001 Device 006: ID 1532:051a Razer USA, Ltd Razer Nari Ultimate
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x051a 
  bcdDevice            1.00
  iManufacturer           4 
  iProduct                5 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x01bd
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0402 Headset
        bAssocTerminal         80
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 3
[... 360 lines cut ...]
          bDescriptorType        34 Report
          wDescriptorLength     426
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1

```

```
Please post the output of
lrwxrwxrwx 1 root root   9 Dec 13 19:38 usb-Razer_Razer_Nari_Ultimate-event-if05 -> ../event6
```

### Packet Captures

https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol -> OK I'll have look, looks like an awesome tuto on USB debugging ! thanks guys

zlooun, 2020-05-27:
Could you add support of this device please?

shawngmc, 2020-07-02:
Do you still need a packet capture for this device? If so, I can try to get one this weekend.

zlooun, 2020-07-03:
@shawngmc
I'm sorry, but not really, after updating windows, I can calmly work with linux via wsl 2.

minecrawler, 2020-07-03:
@zlooun and what about the rest of us on Linux?

hmm-ship-it, 2020-07-06:
@minecrawler This is how things go in open sourced software, it takes time before someone picks up a project. It's all volunteer work.

I tested the Razer Nari Ultimate on Debian Linux 10 64-bit. I installed the open source driver that they provide and the headset works. You are able to hear sound through it.

What it is missing is all the extra configuration for lighting & EQ-like stuff & haptic feedback. This needs to be done through windows still.

gboddin, 2020-07-06:
Hi,

It seems that all you need ( if you don't care about RGB & vibrations ) is a pulseaudio profile :

https://github.com/imustafin/razer-nari-pulseaudio-profile

Regards

z3ntu, 2020-11-17:
Re-opening as the device is still not supported.

pieromarini, 2020-11-17:
Hello, I own a Razer Nari Ultimate **(1532:0051a)**. I know this issue is for the Normal version of the Nari, but since the protocols are the same, the information should be useful anyways.
A few months back I was checking this issue and came up with a simple python program to modify, successfully, the colors and create semi-decent transitions between them.
The script can be found [here](https://gist.github.com/pieromarini/4ec31252b4de76f2181a0467eccfa513). It uses [PyUSB](https://github.com/pyusb/pyusb) to communicate with the device. The script is heavily commented.
From my RE, the protocol is completely different as @imustafin mentioned. There are no more "modes" to set. The device just receives a stream of packets indicating the color it has to set. After that, it stays with that color until it is turned off / disconnected (it then goes to the default green with "Static mode"). 

In the provided script [here](https://gist.github.com/pieromarini/4ec31252b4de76f2181a0467eccfa513#file-nari_ultimate_testing-py-L30), I use a simple lerp to create a transition between two colors and just send them to the device with a small delay in-between [here](https://gist.github.com/pieromarini/4ec31252b4de76f2181a0467eccfa513#file-nari_ultimate_testing-py-L60). Following this logic, and from what I found using packet captures **(found at the bottom)** from a Windows 10 VM, we just need to send these series of packets to create the desired transitions. 

For the case of adjusting brightness, the logic is similar, although it would only be 1 packet sent. For the Nari Ultimate, you can also set On/Off the "vibration" (Haptic Feedback) feature from synapse. This was also just a packet but I'm dumb and didn't save the format it has.

I think this is enough information to build a driver. I don't have much experience writing kernel drivers and I haven't had any spare time to take a look at it recently, but I hope someone can take this information and write it.

<details>
    <summary>Output of <code>lsusb -d 1532:0051a -v</code> for reference</summary>

```
Bus 001 Device 013: ID 1532:051a Razer USA, Ltd Nari Ultimate
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x051a Nari Ultimate
  bcdDevice            1.00
  iManufacturer           4 Razer
  iProduct                5 Razer Nari Ultimate
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x01bd
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0
      iInterface              1 Razer Nari Ultimate - Chat
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0402 Headset
        bAssocTerminal         80
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0
[... 362 lines cut ...]
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     426
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```
</details>

[Packet Captures (includes static, breathing and spectrum RGB modes)](https://github.com/openrazer/openrazer/files/5543470/pcaps.zip)

TermoZour, 2020-12-28:
@nightsky30 how did you add support for Bungee V3? I'm trying to follow you and @999eagle 's work but I'm getting lost in the implementation. I can't even get the driver to read the serial of the headphones correctly. 

Where did you find the transaction ID? I can't seem to find either 1F or 3F in the wireshark pcaps.

nightsky30, 2020-12-29:
@TermoZour Addressing the question of transaction IDs:

I took a look at pieromarini's pcaps (filter Destination by `usb.bus_id == 1 && usb.device_address == 9`), and they seem to have a protocol format which doesn't fit the normal accessory driver. I can see what appear to be the red, green, blue parameters that fluctuate, and there is a header which @pieromarini identified in his linked script. I'm not sure if the normal protocols will also work. Synapse 3 seems to usually send custom frames and doesn't tell us much else.

But a lot of devices support hardware based effects when sent a single packet as well as the custom frames Synapse sends. Since the headset defaults to that static green when unplugged, there might be something there (??)...or it might just default to green using the same strange format.

Here's an example where you can see the razer report for extended matrix static effects and how it sets the transaction ID to 0x3F:
https://github.com/openrazer/openrazer/blob/7c07ba0f7bd9ddcc5a47b818d3bbc0bfece3c292/driver/razerchromacommon.c#L517-L534

Here's a packet capture example from the Mouse Bungee V3 Chroma from Synapse 3 using the transaction ID 0x1F (filter with `usb.bus_id == 1 && usb.device_address == 4`):
[None-Breathing_Single_Green-Toggle.pcapng.zip](https://github.com/openrazer/openrazer/files/5749061/None-Breathing_Single_Green-Toggle.pcapng.zip)

And looking at the pcaps from @999eagle in #949 they seem to also use the same accessory format for custom frames with 0x1F...

One similarity I see is the Nari Ultimate seems to be using an Interface Protocol value of 0 (`bInterfaceProtocol      0`) as do the Kraken Kitty Edition and Mouse Bungee V3 Chroma. That needs to be set properly, or you won't get anything:
https://github.com/openrazer/openrazer/blob/7c07ba0f7bd9ddcc5a47b818d3bbc0bfece3c292/driver/razeraccessory_driver.c#L1067-L1072

## **BUT**

**You may want to try adding the Nari Ultimate to the `Kraken Driver` instead of the `Accessory Driver`. Try modelling after the `Kraken Ultimate`.**

I'm not familiar with that driver, but it also uses the same interface protocol. And it **might** be using the same header values that coincide with what we see in those packet captures pieromarini provided.

If you have a working branch, I'm willing to look at it, but I don't have a headset to test anything. So I can only tell you if you might be missing something that's normally required.

You might get lucky, or it might not work at all. It's really just taking shots in the dark, ruling out what doesn't work. Then moving on from there. If that doesn't work, then it's out of my depth. Maybe @999eagle or @z3ntu might be more help :smile:

TermoZour, 2020-12-29:
I have all my changes on my own fork.

https://github.com/TermoZour/openrazer/tree/nari-ultimate-support

I tried to follow Kraken Ultimate but RazerGenie throws errors and executing list_devices python script throws permission denied errors for pyusb.

nightsky30, 2020-12-29:
@TermoZour Add entries for your device to:
- install_files/appstream/io.github.openrazer.openrazer.metainfo.xml
- install_files/udev/99-razer.rules

See if that helps :smile:

TermoZour, 2020-12-29:
@nightsky30 the udev rules did change something. Can't believe I forgot about that. Those were the reason I got permission errors. The ones from appstream aren't relevant for the driver.

After a full reboot, the driver sees the headphones, I can also see them when executing `examples/list_devices.py`. They are also picked up by RazerGenie but then everything else doesn't work. 

This is the report from the `list_devices.py` script:
```
Found 3 Razer devices

Razer Nari Ultimate:
   type: headset
   serial: XX01
   firmware version: v9.99
   driver version: 2.9.0

Razer Naga Trinity:
   type: mouse
   serial: PM1751H07704103
   firmware version: v1.2
   driver version: 2.9.0

Razer Cynosa Chroma:
   type: keyboard
   serial: PM1743F17804561
   firmware version: v1.0
   driver version: 2.9.0
```

Serial is wrong, firmware is wrong and the effects don't work. They show up in RazerGenie though: **static**, **spectrum**, all 3 **breath variants** and **none**.

Commit here: https://github.com/TermoZour/openrazer/commit/6951ff701651433d3aebc3f921e7cc7c657b3f9e

nightsky30, 2020-12-29:
Look the same here, unfortunately:
[image]

TermoZour, 2020-12-29:
Also, now that the permissions are back working, I can confirm that @pieromarini 's script works and indeed changes the colors.

TermoZour, 2021-01-03:
I've written an attempt to send a predefined USB packet to the headphones from openrazer. They didn't work. Here's the commit I made with the changes. More details are in the commit description. https://github.com/TermoZour/openrazer/commit/3196cf126589df7e56b3a2f2e6ba6873d2b30abe

pieromarini, 2021-01-04:
> What I've also noticed is that the packets from @pieromarini are different from mine. Maybe I'm registering them wrong?
> 
> I'm using a WIndows 10 VM with the headphones passed through and Wireshark running on the VM.

I used the same exact same setup to capture the packets, just FYI.

AgentLoneStar007, 2021-12-01:
Just curious, is this still a work in progress? If the Razer Nari Ultimate isn't supported on OpenRazer, does anyone know other drivers I could use on Arch?

AgentLoneStar007, 2022-06-09:
Just gonna put this here. I am using the Nari Ultimate on EndeavourOS(Arch) Linux. The audio works perfectly, and I even get haptic feedback due to the actual haptic control built into the headset. I do wish, though, that I could change the lighting for the device. Even a static color would rock. Honestly, I don't care about the split game/chat audio feature, or adjusting the haptics, because I can do that in Windows when needed.

senid231, 2022-09-05:
hello, any news about Razer Nari Ultimate support?

I just bought it and try to figure out how can I see the battery level on Linux.

TermoZour, 2022-09-05:
> hello, any news about Razer Nari Ultimate support?
> 
> I just bought it and try to figure out how can I see the battery level on Linux.
> 

As of right now, I am not aware of a solution to check battery percentage from Linux. Not even synapse on Windows shows a battery percentage...

minecrawler, 2022-09-05:
> Not even synapse on Windows shows a battery percentage...

Not a percentage, but an icon with the charge. I don't know how exact it is, but there should be at least rough battery data.
[image]

It feels like my Nari will break before it gets support here, though, as was the case with other hardware. It's sad, but I guess no one wants to add this device.

Rycia, 2022-09-12:
I would also like support.

dreigon2474, 2022-09-18:
> As far as Linux is concerned, the Nari Ultimates are completely unsupported and the audio drivers only function in mono instead of stereo, and no program yet has the capability of supporting it's RGB either.

Not really. There is https://github.com/imustafin/razer-nari-pulseaudio-profile which fixes the audio being in mono. The rest of the stuff is down to someone adding support for the Nari protocols in openrazer

mrquantumoff, 2022-09-18:
> Not really. There is https://github.com/imustafin/razer-nari-pulseaudio-profile which fixes the audio being in mono. The rest of the stuff is down to someone adding support for the Nari protocols in openrazer

I have also forked the project a while ago and applied pipewire fixes + packaged it for Fedora Linux (https://github.com/mrquantumoff/razer-nari-pipewire-profile)

dreigon2474, 2022-09-20:
> I have also forked the project a while ago and applied pipewire fixes + packaged it for Fedora Linux (https://github.com/mrquantumoff/razer-nari-pipewire-profile)

Please correct me if I'm wrong but doesn't the original repo support pipewire? I just want to know the difference as I might want to use this profile for my own naris as I use pipewire

TermoZour, 2022-09-20:
> Please correct me if I'm wrong but doesn't the original repo support pipewire?

It has been updated with pipewire support. I'm using it no problem on Pop!_OS 22.04 LTS.

TermoZour, 2022-09-20:
> As far as Linux is concerned, the Nari Ultimates are completely unsupported and the audio drivers only function in mono instead of stereo

As others mentioned, there is a repo with a very easy install script you can use to get the Nari (or Nari Ultimate) working with proper audio channels.

In terms of RGB, there is a script made by someone arounde here (I can't find it anymore, check one of the issues) and by simply sending a package via USB, the RGB would change and stay that way until you turn the headphones off. Simply modifying that script so you set a custom color (or keyframe an animation) and you're good on that front.

dreigon2474, 2022-09-20:
> In terms of RGB, there is a script made by someone arounde here (I can't find it anymore, check one of the issues) and by simply sending a package via USB, the RGB would change and stay that way until you turn the headphones off. Simply modify that script so you set a custom color (or keyframe an animation) and you're good on that front.

I have already managed to find the script In this thread. Given the scripts relative simplicity I don't see why it is such an issue for the Devs to implement Nari support. Such a shame.

mrquantumoff, 2022-09-21:
> Please correct me if I'm wrong but doesn't the original repo support pipewire? I just want to know the difference as I might want to use this profile for my own naris as I use pipewire

Yes it does, but pipewire isn't packaged for AUR (basically that was the reason I forked it), also after I packaged it for AUR I also packaged it for fedora.

felixZmn, 2022-11-06:
I have the razer nari ultimate too. 

What information is needed to extend the driver support for the device? 
If needed I can provide pcaps

I would implement the support by myself, but my c-skills are probably not good enough for that

z3ntu, 2022-11-09:
@felixZmn You don't really need any C skills, check recent "Add support for " commits to see kind of what needs to be done. Also check https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md

felixZmn, 2022-11-11:
@z3ntu I have one question: I managed to write and load a "stub":

```c
static int connect_rnu(struct hid_device *hdev, const struct hid_device_id *id) {
    printk(KERN_ALERT "razer nari connect called");
    return 0;
}

static void remove_rnu(struct hid_device *hdev) {
    printk(KERN_ALERT "razer nari disconnect called");
}

static const struct hid_device_id razer_nari_ultimate[] = {
        {HID_USB_DEVICE(USB_VENDOR_ID_RAZER, USB_DEVICE_RAZER_NARI_ULTIMATE)},
        {} /* terminating entry */
};

MODULE_DEVICE_TABLE(hid, razer_nari_ultimate);

static struct hid_driver razer_nari_driver = {
        .name = "razer_nari_ultimate",
        .id_table = razer_nari_ultimate,
        .probe = connect_rnu,
        .remove = remove_rnu,
};

module_hid_driver(razer_nari_driver);
```

My Problem now: If I am inserting the usb dongle, the _disconnect_ - message is appearing in /var/logg/syslog. If I am removing the the dongle, the _connect_-message is appeargin... My assumption was, the probe-function is called on connect... what am I missing here?

z3ntu, 2022-11-20:
@felixZmn Sounds weird, probe should be called on USB connect and remove on USB disconnect.

Garfonso, 2022-11-21:
Hi, 

I, too, did switch to Linux and have a Nari Ultimate and would like to see support for it in openrazer. I am also willing to try and add something. 

That said, C skills are a bit rusty, but they should be sufficient, at least I can read the code quite ok. But I never reverse engineered a protocol and must say... that is a bit overwhelming at the beginning. I played a bit with wireshark and synaptic in a VM... I might have understood some things (and I recognized, that for me the [header from that script](https://gist.github.com/pieromarini/4ec31252b4de76f2181a0467eccfa513#file-nari_ultimate_testing-py-L14) is not the same and not static, I did not yet see the 72 at the end. The 05 before that is sometimes there, but sometimes the byte has other values. Also, the FF in the middle is sometimes FD. -> but this might be, because I see all messages and the header is still valid and static for color effects?).

I read all of this thread and the tries in 2020 to get "easy" Nari support by mimicking something else. Given that is two years past now, and a lot did change, wouldn't it be feasible to update and try that again? For example, the accessory approach from @TermoZour ? Or is the protocol still so very much different from other devices?

I did manage to build a modified kernel driver and load it. But somehow I must have missed something from the development documents or so (or the udev rule is not properly installed? I did try to install it using `sudo make udev_install`). If "my" driver is loaded, I get permission errors, that some file in /sys/ is not owned by plugdev (if I correct them, at least the daemon crashes, is that a good sign? :confused: )

TermoZour, 2022-11-21:
@Garfonso I'd say the first step is to just uninstall the openrazer driver from whatever source/package manager you got it and clone the master branch of this repo and install that. 

After you confirm the driver is still working, then you can attempt to modify it bit by bit.

Garfonso, 2023-11-14:
The last few days I did look into this again.

Thanks to the work of other posters before, I managed to write a Nari kernel driver (i.e. it should work for [Nari](https://github.com/openrazer/openrazer/issues/724) and Nari Ultimate)

It made no sense to squeeze this into any of the other drivers. The way of communication seems to be very different. It seems the Device itself does not support *any* modes. Synapse sends color-commands to the device continuously. Even if static color is selected it will send the color every second. That is not necessary, though. Last color & brightness stick. (You can set them in windows and switch to linux, just fine). 
But if you activate something like breathing, synapse will send much more messages and you can see the colors being changed all the time, i.e. synapse does all the work for this device. 

The protocol also seems similar to the one of the very old Man'O War, which has its own issue https://github.com/openrazer/openrazer/issues/43 labled with "unknown lighning protocoll". So I figured, let's add that unknown protocoll.

This is still very basic. The driver basically can set static color and brightness via device files.

I tested a bit around with a dbus monitor and it seems I can control color & brightness currently (for some reason polychromatic does not detect the running daemon... any clues why that is? I run it with this command: `PYTHONPATH="pylib:daemon" python3 ./daemon/run_openrazer_daemon.py -Fv --config=$PWD/daemon/resources/razer.conf --run-dir=/tmp/` -> for some reason I have to run it in /tmp/ because otherwise it complains about file permissions).

Anyway, I'll try to get up some documentation and do a PR soonish. If anyone wants to take a look already, the stuff is currently here:
https://github.com/Garfonso/openrazer/tree/nari-for-itself

Garfonso, 2023-11-16:
Hi,

I have two questions, remaining.

1. I made screenshots from synapse and fiddled more with the protocol (for example there is a energy save feature, where you can configure minutes of inactivity in synape, this then get's converted into seconds and send as two bite Bigendian to the device. This took me some time...). Where can I document this stuff? It this issue the correct place? 

2. There are more functionality in synapse than color & brightness, that I probably can support, like the idle timeout from above or the haptic feedback thing. How to best do that? How does the deamon determine the filenames? For brightness & color I more or less did that by trial & error and looking at other drivers (kraken mostly)... It would be great to get a hint for that from people more experienced with openrazer, especially the python side (python really is not my thing). :-)

StormyIceLeopard, 2024-05-09:
> Hi,
> 
> I have two questions, remaining.
> 
>     .....
> 
>     2. There are more functionality in synapse than color & brightness, that I probably can support, like the idle timeout from above or the haptic feedback thing. .....

Those 2 additional things I do miss. But I also miss the game and chat audio separation. Being able to control the audio of each with the knob on the headset

TermoZour, 2024-05-09:
> But I also miss the game and chat audio separation. Being able to control the audio of each with the knob on the headset

Oh but you can! There is a pipe wire profile made specifically for the Nari.

https://github.com/imustafin/razer-nari-pulseaudio-profile

StormyIceLeopard, 2024-05-09:
> 
> Oh but you can! There is a pipewire profile made specifically for the Nari.
> 
> https://github.com/imustafin/razer-nari-pulseaudio-profile

Oh wow. That actually worked. I use Manjaro, which in turn is using pipewire, so I ended up installing this one https://github.com/mrquantumoff/razer-nari-pipewire-profile via AUR
There is a down side, with both channels maxed out in the audio control. Adjusting the knob slightly results in complete mute. I estimate somewhere between 1/3 and 1/2 of the rotation has the channel with no audio
In other words that I can try to describe it is that it does not use the full rotation of the knob for volume, only a small section. (Yes I know the exact center is 100/100 for both channels, I mean if I start adjusting to lowering the game audio only about a 1/3 or 1/2 of that rotation actually adjusts the audio.) So maybe the word I am looking for is; that it is too sensitive....?

But the fact that it is even possible is amazing! Thank you.

TermoZour, 2024-05-09:
> if I start adjusting to lowering the game audio only about a 1/3 or 1/2 of that rotation actually adjusts the audio.

I am experiencing the same on Linux as well as Windows. I think it's just how the potentiometers are tuned.