openrazer/openrazer #977: Razer Trinity not showing up in latest cca331e

Saksol, 2019-12-17:
### Distribution

Manjaro Juhraya 18.1.4 
kernel 5.4.2-1-MANJARO

Razer trinity is not working yet on latest cca331e using aur openrazer meta git
sudo journalctl -b | grep -i razer_mount  shows nothing

sudo dkms install openrazer-driver/2.6.0
Module openrazer-driver/2.6.0 already installed on kernel 5.4.2-1-MANJARO/x86_64

Daemon is running

lsusb | grep '1532:'                       
Bus 001 Device 002: ID 1532:0067 Razer USA, Ltd Razer Naga Trinity

cat ~/.local/share/openrazer/logs/razer.log shows 

019-12-17 06:34:30 | razer                          | INFO     | Initialising Daemon (v2.6.0). Pid: 1459
2019-12-17 06:34:30 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeMid2019Mercury ---------------- (1532:0245)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth2019 ------------------- (1532:0239)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerCynosaChroma ----------------------- (1532:022A)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdder2013 --------------------- (1532:0037)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdder3500 --------------------- (1532:0054)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdder3_5G --------------------- (1532:0016)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderEssential ---------------- (1532:006E)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerGoliathus -------------------------- (1532:0C01)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerGoliathusExtended ------------------ (1532:0C02)
[... 14 lines cut ...]
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2019-12-17 06:34:30 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
[... cut ...]
2019-12-17 06:34:30 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2019-12-17 06:34:30 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2019-12-17 06:34:30 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2019-12-17 06:34:30 | razer                          | INFO     | Serving DBus
2019-12-17 06:34:30 | razer                          | DEBUG    | DBus called get_serial_list

z3ntu, 2019-12-17:
What's the output of `pacman -Q openrazer-daemon` ?

Saksol, 2019-12-17:
pacman -Q openrazer-daemon
openrazer-daemon 2.6.0-2

z3ntu, 2019-12-17:
If you want to install the latest version of openrazer from git master, you also have to install the -git subpackages (openrazer-daemon-git, openrazer-driver-dkms-git, python-openrazer-git) and not just openrazer-meta-git :)

Saksol, 2019-12-17:
Did then all 4 manually instead of using https://aur.archlinux.org/packages/openrazer-meta-git/ and now it works.
thanks

closed 2019-12-17 by z3ntu