openrazer/openrazer #979: Kernel Module not found in directory (Fedora 31)

Zymion6, 2019-12-18:
### Distribution

Fedora 31
(5.3.7-301.fc31.x86_64)

/var/lib/dkms/openrazer-driver/2.6.0
Ornata Chroma; Abyssus V2

### Support

- [Y] Have you tried rebooting?
- [Y] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I've tried using Polychromatic and Razergenie, but neither seem to work.
I've gone through the troubleshooting guide and the only thing that seems to be at fault is this:

[N0pe@localhost-live ~]$ sudo modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.3.7-301.fc31.x86_64

Pretty sure this is just a Fedora problem.

z3ntu, 2019-12-18:
What happens when you run `sudo dkms install openrazer-driver/2.6.0` ?

Zymion6, 2019-12-18:
Error! echo
Your kernel headers for kernel 5.3.7-301.fc31.x86_64 cannot be found at
/lib/modules/5.3.7-301.fc31.x86_64/build or /lib/modules/5.3.7-301.fc31.x86_64/source.

Zymion6, 2019-12-18:
However I have already run:
 $ dnf install kernel-devel

z3ntu, 2019-12-19:
If you have `kernel-devel` version `5.3.7-301.fc31.x86_64` installed, you should have the folder `/usr/src/kernels/5.3.7-301.fc31.x86_64`, is that correct?

Zymion6, 2019-12-19:
I ran ```dnf update``` yesterday and ended up with ```5.3.16-300.fc31.x86_64``` installed in that folder.
This seems to have fixed the problem as the devices are now being recognized.
Thanks for you help :)

z3ntu, 2019-12-19:
No problem!

closed 2019-12-19 by z3ntu