openrazer/openrazer #980: No control of side LED strip on Lancehead TE in v2.6.0

andrew-wja, 2019-12-19:
### Distribution

Arch Linux

### Support

- [Y] Have you tried rebooting?
- [Y] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Although there is a closed issue https://github.com/openrazer/openrazer/issues/551 which seems to indicate support for this device including side LED strips was merged, side LED strips are still not controllable in v2.6.0

nightsky30, 2019-12-20:
The support within the openrazer daemon was merged.  Support to change settings within the front end is not available in either of the current stable releases of RazerGenie or Polychromatic.  You can use z3ntu's [razer_test](https://github.com/z3ntu/RazerGenie/tree/razer_test) branch, or either of my tweaked branches for [RazerGenie](https://github.com/nightsky30/RazerGenie/tree/side-leds) or [Polychromatic](https://github.com/nightsky30/polychromatic/tree/side-leds).

andrew-wja, 2019-12-20:
Awesome, thank you!

t0msk, 2019-12-28:
Same issue https://github.com/polychromatic/polychromatic/issues/225, when it will be merged into polychromatic?

closed 2020-05-30 by stale[bot]