openrazer/openrazer #982: No Razer product connected on Lancehead

t0msk, 2019-12-28:
### Distribution

Arch Linux 5.4.6-arch1-1

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hello, I have Razer Lancehead Tournament Edition, but polychromatic controller says that there is no No Razer product connected. How can I fix it?

lah7, 2019-12-28:
What's the product's VID/PID?

    lsusb

Polychromatic says "No Razer Product Connected" when the OpenRazer daemon hasn't registered the device **and** was not able to find Razer's vendor ID (1532) from `lsusb` to determine if it's actually connected to the system.

t0msk, 2019-12-28:
I get:
`bash: lsusb: command not found`
So do I need lsusb?

lah7, 2019-12-28:
`lsusb` isn't needed, it's just to double check it's definitely the same hardware. Kinda suspected `lsusb` wasn't installed.

"Device not recognised" commonly down to an improperly installed/loaded driver. What's the output of:

    sudo modprobe razermouse

And:

    tail ~/.local/share/openrazer/logs/razer.log

t0msk, 2019-12-28:
```
[tomsk@tomsk-PC ~]$ sudo modprobe razermouse                    
modprobe: FATAL: Module razermouse not found in directory /lib/modules/5.4.6-arch1-1
[tomsk@tomsk-PC ~]$ tail -f ~/.local/share/openrazer/logs/razer.log
2019-12-28 13:25:49 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2019-12-28 13:25:49 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2019-12-28 13:25:49 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2019-12-28 13:25:49 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2019-12-28 13:25:49 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2019-12-28 13:25:49 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2019-12-28 13:25:49 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2019-12-28 13:25:49 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2019-12-28 13:25:49 | razer                          | INFO     | Serving DBus
2019-12-28 13:25:50 | razer                          | DEBUG    | DBus called get_serial_list
2019-12-28 13:40:24 | razer                          | DEBUG    | DBus called get_serial_list
2019-12-28 13:40:27 | razer                          | INFO     | Stopping daemon.
2019-12-28 13:40:30 | razer                          | INFO     | Initialising Daemon (v2.6.0). Pid: 3868
2019-12-28 13:40:30 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
2019-12-28 13:40:30 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
[... cut ...]
2019-12-28 13:40:30 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2019-12-28 13:40:30 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2019-12-28 13:40:30 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2019-12-28 13:40:30 | razer                          | INFO     | Serving DBus
2019-12-28 13:40:31 | razer                          | DEBUG    | DBus called get_serial_list
^C
[tomsk@tomsk-PC ~]$ 

```

lah7, 2019-12-28:
The DKMS module isn't installed, make sure you have your Linux kernel headers installed (`linux-headers`) and try installing the OpenRazer driver again:

    sudo dkms install -m openrazer-driver/2.6.0

Once successfully installed, reboot. If things still don't work, then it's likely [secure boot](https://github.com/openrazer/openrazer/wiki/Troubleshooting#kernel-modules-loading) needs to be disabled to allow the unsigned driver to load.

t0msk, 2019-12-28:
I installed openrazer from AUR, and this is output from your command:
```
[tomsk@tomsk-PC ~]$ sudo dkms install -m openrazer-driver/2.6.0
[sudo] password for tomsk: 
Error! echo
Your kernel headers for kernel 5.4.6-arch1-1 cannot be found at
/usr/lib/modules/5.4.6-arch1-1/build or /usr/lib/modules/5.4.6-arch1-1/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
```

t0msk, 2019-12-28:
Thanks, it is working now :)

closed 2019-12-28 by t0msk