openrazer/openrazer #984: "OpenRazer daemon is not installed"

johnwilliamsdigital, 2019-12-30:
### Distribution

### Support

- [ ] Have you tried rebooting? Of course
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes.

I'm getting the attached error. 

Not sure what to do to fix it, I'm not an expert programmer so baby steps would be very helpful. 

Kubuntu 19.10

Deathaddereliete Mouse

hedmo, 2020-01-08:
@johnwilliamsdigital 

can you post the output of :+1: 

`lsmod | grep razer`
and 
`openrazer-daemon`

regards hedmo

johnwilliamsdigital, 2020-01-09:
Hey, thanks for replying. 

I got it to run with: 

`openrazer-daemon -Fv
sudo modprobe razerkbd`

However, my purpose was to reassign the thumb buttons, which I realized afterwards this app doesn't do. 

Any reccomendations for reassigning mouse buttons?

Thanks!

hedmo, 2020-01-10:
For botton mapping: https://bues.ch/cms/hacking/razercfg.html

closed 2020-06-08 by stale[bot]