openrazer/openrazer #987: Support for Viper Ultimate

nseguin42, 2020-01-04:
**Device Info**

- Name: Viper Ultimate
- Product Number: RC30-03050100-0000
- Model Number: RC30-030501

**Device Information**
This is the output of `lsusb -d 1532: -v`:
```
Bus 001 Device 002: ID 1532:007b Razer USA, Ltd Razer Viper Ultimate Dongle
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x007b 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

This is the output of `ls -lah /dev/input/by-id/`:
```
total 0
drwxr-xr-x. 2 root root 280 Jan  3 17:50 .
drwxr-xr-x. 4 root root 660 Jan  3 17:50 ..
lrwxrwxrwx. 1 root root   9 Jan  3 17:50 usb-Razer_Razer_Viper_Ultimate_Dongle-event-if01 -> ../event7
lrwxrwxrwx. 1 root root   9 Jan  3 17:50 usb-Razer_Razer_Viper_Ultimate_Dongle-event-mouse -> ../event3
lrwxrwxrwx. 1 root root   9 Jan  3 17:50 usb-Razer_Razer_Viper_Ultimate_Dongle-if01-event-kbd -> ../event4
lrwxrwxrwx. 1 root root   9 Jan  3 17:50 usb-Razer_Razer_Viper_Ultimate_Dongle-if02-event-kbd -> ../event8
lrwxrwxrwx. 1 root root   9 Jan  3 17:50 usb-Razer_Razer_Viper_Ultimate_Dongle-mouse -> ../mouse0
```

zstewart93, 2020-02-10:
Hello all, I have one of these as well if you need testing feedback. Just point me towards the documentation for grabbing the information you need and let me know what you need. Thanks!

JKing-B16, 2020-02-12:
I would love to have this as well!

z3ntu, 2020-02-12:
Instructions on how to make USB packet capture is documented here: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

brokenshakles, 2020-02-29:
The referenced link says the charging dock for the ultimate (wireless) will not be included in this ticket, will that be a future addition?

lukelex, 2020-03-31:
+1 for being a guinea pig

z3ntu, 2020-04-06:
Support is merged and will be a part of the next release!

closed 2020-04-06 by z3ntu