openrazer/openrazer #988: Support for Razer Electra V2 USB

geozal, 2020-01-04:
**Device Info**

    Name: Kraken Tournament Edition

**Device Information**

Bus 001 Device 003: ID 1532:0514 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0514 
  bcdDevice            0.14
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          296
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength           70
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0402 Headset
        bAssocTerminal          0
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 8
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 2
        bSourceID               1
        bControlSize            1
[... 241 lines cut ...]
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0023  1x 35 bytes
        bInterval               1

total 0
drwxr-xr-x 2 root root 220 Jan  4 14:28 .
drwxr-xr-x 4 root root 640 Jan  4 14:28 ..
lrwxrwxrwx 1 root root   9 Jan  4 14:28 usb-099a_Gaming_Keyboard-event-if01 -> ../event4
lrwxrwxrwx 1 root root   9 Jan  4 14:28 usb-099a_Gaming_Keyboard-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Jan  4 14:28 usb-099a_Gaming_Keyboard-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Jan  4 14:28 usb-099a_Gaming_Keyboard-if01-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Jan  4 14:28 usb-099a_Gaming_Keyboard-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Jan  4 14:28 usb-Corsair_Corsair_Gaming_M65_Pro_RGB_Mouse_0F02302CAEB610425614827BF5001946-event-if00 -> ../event9
lrwxrwxrwx 1 root root   9 Jan  4 14:28 usb-Corsair_Corsair_Gaming_M65_Pro_RGB_Mouse_0F02302CAEB610425614827BF5001946-event-joystick -> ../event7
lrwxrwxrwx 1 root root   9 Jan  4 14:28 usb-Corsair_Corsair_Gaming_M65_Pro_RGB_Mouse_0F02302CAEB610425614827BF5001946-joystick -> ../mouse1
lrwxrwxrwx 1 root root   9 Jan  4 14:28 usb-Razer_Razer_Electra_V2_USB_AC1850D06701725-event-if03 -> ../event2

antimech, 2020-01-11:
Why the name is `Kraken Tournament Edition`?

geozal, 2020-01-11:
No idea. Fixed it.

nikoloskid, 2020-03-25:
> No idea. Fixed it.

Did you had microphone problem with it on Linux?

wisner23, 2020-06-17:
> > No idea. Fixed it.
> 
> Did you had microphone problem with it on Linux?

i'm getting problems with my Razer Electra V2 USB, it is randomly closing the audio/mic and switching for default notebook audios

antimech, 2020-06-25:
> > > No idea. Fixed it.
> > 
> > 
> > Did you had microphone problem with it on Linux?
> 
> i'm getting problems with my Razer Electra V2 USB, it is randomly closing the audio/mic and switching for default notebook audios

I'm having a similar issue. It looks like driver/hardware problem. It's not consistent, so I can't reproduce it. Currently It's randomly mutes my sound during gameplay, I have to disconnect it and plug it back it, it also causes some games to crash or having to restart to get the sound back. I'm not sure if it's Windows only problem (wasn't able to reproduce it on Linux yet).