openrazer/openrazer #989: No device was detected (Razer Naga Trinity 	1532:0067)

vlad34, 2020-01-04:
Hi, 

thanks for the software, i have a Razer Naga Trinity '1532:0067' which is not recognized by RazerGenie, RazerCommander and Polychromatic on fedora.

Openrazer is well installed and I added my user in the "plugdev" group.

----------------

Command lines:

sudo dnf config-manager --add-repo https://download.opensuse.org/repositories/hardware:razer/Fedora_31/hardware:razer.repo

sudo dnf install openrazer-meta

sudo gpasswd -a vlad34 plugdev

-----------------

Message from RazerGenie:

No device was detected

The OpenRazer Daemon didn't detect a device that is supported.

This could also be caused due to a misconfiguration of this PC.

----------------

thanks for the help.

[image]

lah7, 2020-01-04:
Hi, Razer Naga Trinity hasn't been released to stable yet (currently **v2.6.0**). It will be [included in the next release (v2.7.0).](https://github.com/openrazer/openrazer/compare/v2.6.0...master) 

Your installation is probably working as expected. If you'd like to build yourself a v2.7.0 package for the time being, [check out these instructions for Fedora](https://github.com/openrazer/openrazer/wiki/Building-a-package#fedora).

vlad34, 2020-01-11:
Hi,

I updated to 2.7.0, the Razer Naga Trinity 1532: 0067 mouse is not recognized.

thank :) .

z3ntu, 2020-01-11:
Can you the command `sudo dkms install openrazer-driver/2.7.0` and `sudo modprobe razerkbd` please?

vlad34, 2020-01-11:
Done, the mouse is recognized ;) .

for information:
"
Creating symlink /var/lib/dkms/openrazer-driver/2.7.0/source ->
                 /usr/src/openrazer-driver-2.7.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
KERNELDIR=/lib/modules/5.4.8-200.fc31.x86_64/build make driver....
cleaning build area...

DKMS: build completed.

razerkbd.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.8-200.fc31.x86_64/extra/

razermouse.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.8-200.fc31.x86_64/extra/

razermousemat.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.8-200.fc31.x86_64/extra/

razerkraken.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.8-200.fc31.x86_64/extra/

razeraccessory.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.8-200.fc31.x86_64/extra/

razercore.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.8-200.fc31.x86_64/extra/
Adding any weak-modules

depmod....

DKMS: install completed.
"

Thank. :)

closed 2020-01-11 by vlad34

sezanzeb, 2020-10-26:
> Can you the command `sudo dkms install openrazer-driver/2.7.0` and `sudo modprobe razerkbd` please?

Thanks! I also had to reboot after running modprobe