openrazer/openrazer #990: Razer Blade Stealth (mid 2017) gets stuck on Fn lighting (just function and arrow keys lit) when using matrix_effect_custom

tommyfao, 2020-01-07:
### Distribution

Fedora 31
Linux localhost.localdomain 5.4.7-200.fc31.x86_64 #1 SMP Tue Dec 31 22:25:12 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

When using custom static keyboard lighting on my Razer Blade Stealth mid-2017, pressing the fn key turns off all key lights, and illuminates only the function row and arrow keys - as expected. However, upon releasing the fn key, the keyboard remains in this state, and does not change until a new lighting scheme is applied. 

I experienced this issue using razergenie, but I am also experiencing the same thing now when writing the frames with my own scripts, which I switched to so that I could easily reapply the frame with a shortcut.

I dual boot with windows and do not have the same issue in windows. I am using Fedora now, but recently was using Ubuntu and had the same problem.

I am using openrazer-daemon 2.6.0. This issue does not apply when use any other effects, only the custom frame.

lezed1, 2020-01-07:
I have the same behavior on my RBS. I think it's just how the keyboard works. If you use the custom mode, it requires you to send a new frame after pressing the fn key. I never found a way to listen to the fn key, so I just sent the same frame once a second if I use a static image.

tommyfao, 2020-01-07:
> I think it's just how the keyboard works.
Do you mean you think it's an issue with the keyboard firmware itself? It doesn't have the same issue on Windows - but perhaps synapse covers it up?

> I never found a way to listen to the fn key, so I just sent the same frame once a second if I use a static image.
Yeah actually that makes more sense than my solution. I'm going to use that thanks!

lezed1, 2020-01-07:
I don't think it's an issue with the firmware, I think it's how it is designed. Synapse pushes frames at some rate continuously if I remember correctly. (I've never tried a custom, static image, but I'd be surprised if it was special cased to be different than a dynamic effect)

tommyfao, 2020-01-07:
Ah I see, that makes sense. I misunderstood the difference between the static effect, and a custom frame. I thought they were essentially the same thing, but I think I understand now - the static effect is "effected" by continually pushing a frame. Whereas pushing a custom frame is just that - a single frame being pushed.

Thank you for clearing that up for me. I appreciate it.

closed 2020-01-07 by tommyfao