openrazer/openrazer #992: logo on/off toggle support for Blade Advanced mid 2019

TheodoreT, 2020-01-08:
I am trying to check whether I can turn on/off the logo of my blade but it seems that OpenRazer currently does not support this option in my rig (Blade Advanced Mid 2019 OLED).

It seems that it is disabled by default according to the following:

```
-------------------------------------------------------------------------------

# Device Info
Name: Razer Blade 15 (Mid 2019) Mercury White
Type: keyboard
 
# Device Capabilities
{'name': True, 'type': True, 'firmware_version': True, 'serial': True, 'brightness': True, 'macro_logic': False, 'keyboard_layout': True, 'lighting': True, 'lighting_breath_single': True, 'lighting_breath_dual': True, 'lighting_breath_triple': False, 'lighting_breath_random': True, 'lighting_wave': True, 'lighting_reactive': True, 'lighting_none': True, 'lighting_spectrum': True, 'lighting_static': True, 'lighting_starlight_single': True, 'lighting_starlight_dual': True, 'lighting_starlight_random': True, 'lighting_ripple': True, 'lighting_ripple_random': True, 'lighting_pulsate': False, 'lighting_led_matrix': True, 'lighting_led_single': False, 'lighting_logo': False, 'lighting_logo_active': False, 'lighting_logo_blinking': False, 'lighting_logo_brightness': False, 'lighting_logo_pulsate': False, 'lighting_logo_spectrum': False, 'lighting_logo_static': False, 'lighting_logo_none': False, 'lighting_logo_reactive': False, 'lighting_logo_wave': False, 'lighting_logo_breath_single': False, 'lighting_logo_breath_dual': False, 'lighting_logo_breath_random': False, 'lighting_scroll': False, 'lighting_scroll_active': False, 'lighting_scroll_blinking': False, 'lighting_scroll_brightness': False, 'lighting_scroll_pulsate': False, 'lighting_scroll_spectrum': False, 'lighting_scroll_static': False, 'lighting_scroll_none': False, 'lighting_scroll_reactive': False, 'lighting_scroll_wave': False, 'lighting_scroll_breath_single': False, 'lighting_scroll_breath_dual': False, 'lighting_scroll_breath_random': False, 'lighting_left': False, 'lighting_left_active': False, 'lighting_left_brightness': False, 'lighting_left_spectrum': False, 'lighting_left_static': False, 'lighting_left_none': False, 'lighting_left_reactive': False, 'lighting_left_wave': False, 'lighting_left_breath_single': False, 'lighting_left_breath_dual': False, 'lighting_left_breath_random': False, 'lighting_right': False, 'lighting_right_active': False, 'lighting_right_brightness': False, 'lighting_right_spectrum': False, 'lighting_right_static': False, 'lighting_right_none': False, 'lighting_right_reactive': False, 'lighting_right_wave': False, 'lighting_right_breath_single': False, 'lighting_right_breath_dual': False, 'lighting_right_breath_random': False, 'lighting_backlight': False, 'lighting_backlight_active': False, 'game_mode_led': False, 'macro_mode_led': False, 'macro_mode_led_effect': False, 'macro_mode_modifier': False}

-------------------------------------------------------------------------------
```

since `lighting_logo` and `lighting_logo_active` are set to `false`.  

I tried to follow the suggestion from here https://github.com/polychromatic/polychromatic/issues/104#issuecomment-571785769 but this does not seem to work. Thus, I would like to ask whether you have any other suggestion.

z3ntu, 2020-01-10:
Hi, you can try adding the following line to the `case` statement for your device: https://github.com/openrazer/openrazer/blob/94ae491/driver/razerkbd_driver.c#L2017 (your device should be `USB_DEVICE_ID_RAZER_BLADE_MID_2019_MERCURY`). 

Then after rebuilding the kernel driver (and loading the new kernel driver) there should be a file under `/sys/bus/hid/drivers/razerkbd/0*/logo_led_state` where you can use the following https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver#logo_led_state---sets-the-onoff-state-of-the-blade-stealth-lid-logo-write-only to try to toggle the logo on and off.

If you need more detailed help, feel free to ask :)

TheodoreT, 2020-01-20:
@z3ntu thanks this worked. Do you know how I can enable the toggle through the polychromatic interface. I tried to do what I was suggested here https://github.com/polychromatic/polychromatic/issues/104#issuecomment-571785769 but it didn't really work. Otherwise, I will ask back there again.

commit 012ac7fc (2020-01-23): Add support for toggling logo light on Razer Blade Mid 2019

Fixes #992
closed 2020-01-23 by z3ntu