openrazer/openrazer #993: Razer naga trinity

woomdawg, 2020-01-09:
Naga trinity not recognized. I have rebooted but still not recognized. Ubuntu 19.10

z3ntu, 2020-01-09:
Can you please run the command `sudo modprobe razerkbd` and the command `sudo dkms install openrazer-driver/2.6.0` and see if that works?

woomdawg, 2020-01-10:
I did that and it did not work. This the terminal from those commands. 
scott@scott-OEM:~$ sudo modprobe razerkbd
[sudo] password for scott: 
scott@scott-OEM:~$ sudo modprobe razerkbd
scott@scott-OEM:~$ 1655
1655: command not found
scott@scott-OEM:~$ sudo dkms install
Error! Invalid number of arguments passed.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>
scott@scott-OEM:~$ sudo dkms install openrazer-driver/2.6.0
Module openrazer-driver/2.6.0 already installed on kernel 5.3.0-26-generic/x86_64
scott@scott-OEM:~$

z3ntu, 2020-01-10:
The Razer Naga Trinity hasn't been released to stable yet (currently **v2.6.0**). It will be [included in the next release (v2.7.0).](https://github.com/openrazer/openrazer/compare/v2.6.0...master) 

Your installation is probably working as expected. If you'd like to build yourself a v2.7.0 package for the time being, [check out these instructions for Ubuntu](https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint).

closed 2020-01-10 by z3ntu