openrazer/openrazer #994: Support for [device name here]

valterleonardo, 2020-01-09:
### Device Info

* Name: Razer Cyclosa
* Product Number: RZ03-00410200
* Model Number: RZ03-0041, Gaming keyboard

### Device Information
```
Please post the output of
lsusb -d 1532: -v

Bus 003 Device 003: ID 1532:0110 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0110 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
```

```
Please post the output of
ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 200 jan  9 14:24 .
drwxr-xr-x 4 root root 420 jan  9 14:24 ..
lrwxrwxrwx 1 root root   9 jan  9 14:24 usb-Plantronics_Plantronics_C310-M_CF16FAB2DA114D5DB1AAB87F2085FD63-event-if03 -> ../event8
lrwxrwxrwx 1 root root  10 jan  9 14:24 usb-Razer_Razer_Abyssus_1800-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 jan  9 14:24 usb-Razer_Razer_Abyssus_1800-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 jan  9 14:24 usb-Razer_Razer_Abyssus_1800-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 jan  9 14:24 usb-Razer_Razer_Cyclosa-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 jan  9 14:24 usb-Razer_Razer_Cyclosa-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 jan  9 14:24 usb-Razer_Razer_Cyclosa-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 jan  9 14:24 usb-Razer_Razer_Cyclosa-if02-mouse -> ../mouse1
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

z3ntu, 2020-01-10:
Hi, please read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we need packet captures of the Windows software (Razer Synapse likely). Thanks!

valterleonardo, 2020-01-11:
> Hi, please read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we need packet captures of the Windows software (Razer Synapse likely). Thanks!

I'll do it as soon as possible

Zeverity, 2021-12-28:
```
lsusb -d 1532: -v

Bus 004 Device 003: ID 1532:0020 Razer USA, Ltd Razer Abyssus 1800
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0110 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
```
```
ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 200 Dec 28 04:14 .
drwxr-xr-x 4 root root 600 Dec 28 04:14 ..
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Abyssus_1800-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Abyssus_1800-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Abyssus_1800-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-event-if01 -> ../event4
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-if02-mouse -> ../mouse0
```

Zeverity, 2021-12-28:
```
lsusb -d 1532: -v

Bus 004 Device 003: ID 1532:0020 Razer USA, Ltd Razer Abyssus 1800
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0020 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Abyssus 1800
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
[... 133 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
```
```
ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 200 Dec 28 04:14 .
drwxr-xr-x 4 root root 600 Dec 28 04:14 ..
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Abyssus_1800-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Abyssus_1800-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Abyssus_1800-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-event-if01 -> ../event4
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Dec 28 04:14 usb-Razer_Razer_Cyclosa-if02-mouse -> ../mouse0
```

Zeverity, 2021-12-31:
**I've this keyboard and I use Linux. Please, support for the Razer Cyclosa. <3**