openrazer/openrazer #995: M1 Macro key opens GNOME Settings

TheNyneR, 2020-01-12:
### Distribution

Distributor ID:	Ubuntu
Description:	Pop!_OS 19.10
Release:	19.10
Codename:	eoan

### Support

- [X ] Have you tried rebooting?
- [X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Using Razer BlackWidow Chroma V2
When pressing the "M1" key the GNOME settings are opened. Trying to record a macro for this key doesn't change that behavior. Checking which key gets pressed it states the "F13" key is pressed, even though the keyboard does not have this key.

z3ntu, 2020-01-12:
I think this is (at least technically) related to #579 . In short: The driver uses F13 to "expose" the M* buttons because there aren't any other key events produced otherwise. See the source: https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L1892

TheNyneR, 2020-01-20:
I found a workaround now.

For anyone having the same problem:
You can use "dconf-editor" or a similar utility to disable "control-center-static" in org > gnome > settings-daemon > plugins > media-keys.

closed 2020-01-20 by TheNyneR