openrazer/openrazer #1004: Support for [device name here]

CalcProgrammer1, 2020-01-23:
### Device Info

* Name: Razer Nommo Pro
* Product Number:
* Model Number:

### Device Information
Razer 2.1 THX-certified speaker system.  Each speaker has 8 RGB LEDs around the base.

USB PID:VID - 1532:0518

I'm experimenting with adding support for the Nommo Pro to the razeraccessory driver.  My Nommo Pro is actually a pre-production prototype that Razer sent me for Keyboard Visualizer development and is a bit different from the released hardware, but it shows up as the release unit on Synapse 3.  Once I figure out all the commands I will open a PR.

I also have the Razer Nommo Chroma, also a pre-production prototype but it looks almost identical to the released version.

Modes:
* Custom - Not working
* None - Working, use razer_chroma_extended_matrix_effect_none(VARSTORE, ZERO_LED);
* Static - Not working
* Breathing - Working, use razer_chroma_extended_matrix_effect_breathing_random(VARSTORE, ZERO_LED); (and _single, _dual)
* Spectrum - Working, use razer_chroma_extended_matrix_effect_spectrum(VARSTORE, ZERO_LED);
* Wave - Not working
* Reactive - Not tested

CalcProgrammer1, 2020-01-23:
The Nommo Chroma is 1532:0517 and the mode compatibility is the same.  It has 2 rows of 24 columns.  The left and right speakers are reversed from the Pro.

CalcProgrammer1, 2020-03-24:
Opened a pull request to add both the Nommo Pro and Nommo Chroma:

https://github.com/openrazer/openrazer/pull/1061

z3ntu, 2020-05-02:
Fixed with #1061

closed 2020-05-02 by z3ntu