openrazer/openrazer #1011: Error

Alexandre6660, 2020-01-27:
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-01-27 01:14:08 -03; 24s ago
     Docs: man:openrazer-daemon(8)
  Process: 3720 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 3720 (code=exited, status=1/FAILURE)
      CPU: 231ms

jan 27 01:14:08 localhost.localdomain systemd[1154]: Starting Daemon to manage razer devices in userspace...
jan 27 01:14:08 localhost.localdomain openrazer-daemon[3720]: Starting daemon.
jan 27 01:14:08 localhost.localdomain openrazer-daemon[3720]: 2020-01-27 01:14:08 | razer                          | CRITICAL | User is not a member of the plugdev group
jan 27 01:14:08 localhost.localdomain openrazer-daemon[3720]: Stopping daemon.
jan 27 01:14:08 localhost.localdomain systemd[1154]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
jan 27 01:14:08 localhost.localdomain systemd[1154]: openrazer-daemon.service: Failed with result 'exit-code'.
jan 27 01:14:08 localhost.localdomain systemd[1154]: Failed to start Daemon to manage razer devices in userspace.

Alexandre6660, 2020-01-27:
[image]

Alexandre6660, 2020-01-27:
[image]

lah7, 2020-01-27:
> CRITICAL | User is not a member of the plugdev group

Please run this command, log out and log back in to the computer:

    sudo gpasswd -a $USER plugdev

Alexandre6660, 2020-01-27:
Before you send it I found this command in another forum and it worked, thanks anyway! But now I have another problem, I change the dpi but I don't feel the effects, it seems that it just doesn't change. :(

Alexandre6660, 2020-01-27:
[image]

z3ntu, 2020-01-27:
So you mean when you untick the "Set Logo Active" checkbox, then the light on the logo LED stay on?

Alexandre6660, 2020-01-28:
No, all functions work except for dpi.

Alexandre6660, 2020-01-28:
I tried to reinstall using the flatpack and this error appeared.

Alexandre6660, 2020-01-28:
[image]

closed 2020-10-04 by stale[bot]