openrazer/openrazer #1012: Support for Mouse Deathadder V2

sachanganesh, 2020-01-27:
### Device Info

* Name: DeathAdder v2
* Product Number: RZ01-03210100
* Model Number: RZ01-0321

### Device Information
```
Bus 003 Device 010: ID 1532:0084 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0084 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          109
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     325
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
```

```
Please post the output of
    total 0
drwxr-xr-x 2 root root 300 Jan 25 17:12 .
drwxr-xr-x 4 root root 660 Jan 25 17:12 ..
lrwxrwxrwx 1 root root   9 Jan 25 15:44 usb-04d9_daskeyboard-event-if01 -> ../event9
lrwxrwxrwx 1 root root   9 Jan 25 15:44 usb-04d9_daskeyboard-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 Jan 25 15:44 usb-13ba_Generic_USB_K_B-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Jan 25 15:44 usb-13ba_Generic_USB_K_B-if01-event-mouse -> ../event11
lrwxrwxrwx 1 root root   9 Jan 25 15:44 usb-13ba_Generic_USB_K_B-if01-mouse -> ../mouse2
lrwxrwxrwx 1 root root   9 Jan 25 15:44 usb-HP_HP_Wireless_Comfort_Desktop-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Jan 25 15:44 usb-HP_HP_Wireless_Comfort_Desktop-if01-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 Jan 25 15:44 usb-HP_HP_Wireless_Comfort_Desktop-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Jan 25 15:44 usb-No_brand_SP02-A1-event-kbd -> ../event12
lrwxrwxrwx 1 root root   9 Jan 25 17:12 usb-Razer_Razer_DeathAdder_V2-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 Jan 25 17:12 usb-Razer_Razer_DeathAdder_V2-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Jan 25 17:12 usb-Razer_Razer_DeathAdder_V2-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Jan 25 17:12 usb-Razer_Razer_DeathAdder_V2-mouse -> ../mouse1
```

### Packet Captures

Don't have time right at this moment; please let me know if it's necessary and I'll try to find the time to give this information.

### Synapse Screenshots

Same as above.

---

Thanks for your help! Please let me know if the lack of packet info is holding back progress.

json-born, 2020-02-06:
+1 any updates on this?

DaMightyZombie, 2020-03-01:
I've gathered a few Synapse screenshots, working on pcaps now. Please tell me if these images are insufficient.
[image]
[image]
[image]
[image]

dobsonaron, 2020-03-27:
I just bought this mouse and it's awesome! I would like to support the development, so please let me know what is missing and I'll try to provide it!

Thank you very much for your time in advance!

lsiudut, 2020-04-09:
Just made my DeathAdder V2 work. I'll put up a PR later today up for review. There's a bit of dog science in there.

lsiudut, 2020-04-10:
https://github.com/openrazer/openrazer/pull/1093

I need some feedback from maintainers of the code though as there's one huge caveat (described in the PR). It will most likely require another iteration. This code works though, feel free to test it.

Be warned that it introduces some devices filtering that may make other Razer devices not work. This is exactly the part that needs some experienced eyes. I have no other hardware of Razer to test it so it's pure dog science.

z3ntu, 2020-05-02:
Fixed with #1093

closed 2020-05-02 by z3ntu