openrazer/openrazer #1015: Support for Lancehead Wireless (2019)

mduft, 2020-01-29:
### Device Info

* Name: Razer Lancehead Wireless
* Product Number: RZ01-02570100
* Model Number: RC30-025701

### Device Information
```
# lsusb -d 1532: -v

Bus 001 Device 029: ID 1532:006f Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x006f 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Lancehead Wireless
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

```
# ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 240 Jan 27 14:28 .
drwxr-xr-x 4 root root 700 Jan 27 14:28 ..
lrwxrwxrwx 1 root root  10 Dec 12 07:33 usb-046d_HD_Pro_Webcam_C920_8C3FAABF-event-if00 -> ../event15
lrwxrwxrwx 1 root root  10 Dec 12 07:33 usb-Microsoft_Natural®_Ergonomic_Keyboard_4000-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Dec 12 07:33 usb-Microsoft_Natural®_Ergonomic_Keyboard_4000-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 Jan 27 14:28 usb-Razer_Razer_Lancehead_Wireless-event-if01 -> ../event24
lrwxrwxrwx 1 root root   9 Jan 27 14:28 usb-Razer_Razer_Lancehead_Wireless-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 Jan 27 14:28 usb-Razer_Razer_Lancehead_Wireless-if01-event-kbd -> ../event22
lrwxrwxrwx 1 root root  10 Jan 27 14:28 usb-Razer_Razer_Lancehead_Wireless-if02-event-kbd -> ../event26
lrwxrwxrwx 1 root root   9 Jan 27 14:28 usb-Razer_Razer_Lancehead_Wireless-mouse -> ../mouse2
lrwxrwxrwx 1 root root   9 Dec 12 07:33 usb-SteelSeries_SteelSeries_Arctis_7-event-if05 -> ../event8
lrwxrwxrwx 1 root root  10 Dec 12 07:33 usb-SunplusIT_Inc_Integrated_Camera-event-if00 -> ../event16
```

### Packet Captures

Unfortunately our network (company) does not allow installation of synapse on my Windows VM, so I cannot create pcaps :(

### Synapse Screenshots
I have the same mouse at home (but no linux there), so I can create synapse screenshots at least later on.

DensityFunction, 2020-02-18:
I have both linux (ubuntu) and windows (no VM) and the mouse, unfortunately i dont know to create pcaps i will create it if you give me a tutorial to follow.

z3ntu, 2020-02-18:
@DensityFunction You should be able to follow https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol for this.

nightsky30, 2020-03-22:
@z3ntu I can provide pcaps if that would be helpful, but I checked brightness and it appears to be spamming packets.

I actually bought this device and have a mostly functioning branch.  What is odd is that it sets everything just fine (effects, dpi, poll rate, brightness), but the only way I've gotten it to read anything is by modifying the RAZER_MOUSE_WAIT_MIN_US, RAZER_MOUSE_WAIT_MAX_US values to have a wider range.

The good news it that it appears to be reading all the correct values when it does.  Do you have any recommendations on the max and min waits?  I can continue to modify them to see what the smallest range outside of 600-800 works reliably with this mouse.

nightsky30, 2020-03-23:
@mduft @DensityFunction  I submitted a pull request.  The mouse is working very well for me (except battery status).  Feel free to test if you wish.

mduft, 2020-03-23:
> @mduft @DensityFunction I submitted a pull request. The mouse is working very well for me (except battery status). Feel free to test if you wish.

Thats great new, thanks! I will test it immediately once I'm allowed to go back to work :D My mouse is at the company...

z3ntu, 2020-03-23:
@nightsky30  For the battery, yes, pcaps would be very helpful for that. Thanks for the PR!

nightsky30, 2020-03-24:
Accessing the Customize screen for the Lancehead Wireless mouse from the "Synapse" overview type screen.  This loads the 100% battery indicator in the upper right hand corner among other things.  I hope this generates the packets required that indicate battery level. I'm not sure how they would be generated otherwise :thinking: except maybe running the mouse battery down:
[image]

The Power" screen and it's features:
[image]

[lancehead_power_pcaps.zip](https://github.com/openrazer/openrazer/files/4372633/lancehead_power_pcaps.zip)

z3ntu, 2020-03-24:
pcaps are fine :)

But (un)fortunately the charging commands look the same as on the other devices. The only thing that I notice is that the packets use the transaction id 0x1f and not 0x3f as you've used in your PR. "get charging status" is data size 0x02, class 0x07 cmd 0x84 with args 0x0000 (and battery level is the same but cmd 0x80 and the response in the pcaps is 0x00ff)

nightsky30, 2020-03-24:
> pcaps are fine :)
>
> But (un)fortunately the charging commands look the same as on the other devices. The only thing that I notice is that the packets use the transaction id 0x1f and not 0x3f as [...]

Cool!  Those match the default razer_report values returned in razerchromacommon.c, so then I guess I should just need to test with tansaction ID 1f ??

PR #1055 (2020-05-02): Add support for Razer Lancehead Wireless (2019)
Add support for Lancehead Wireless 2019 fixes #1015 

 - Adds support for Lancehead Wireless
 - Fixes reading settings/values for Lancehead Wireless, Mamba Wireless, and perhaps other mice
 - Fixes reading battery status for Lancehead Wireless

Everything has been tested wired and wirelessly including setting all effects, brightness, dpi, pollrate, and reading all values including firmware and serial.

Two issues:

1) This mouse appears to be very strange about timing when reading values.

The only way to read ANY values was to dramatically incre [... cut]
closed 2020-05-02 by z3ntu