openrazer/openrazer #1018: Support for Razer Mouse Dock

laurinneff, 2020-01-31:
### Device Info

* Name: Razer Mouse Dock
* Product Number: RC30-03050200-0000
* Model Number: RC30-030502

### Device Information
```
(I'll add when I reboot into Linux)
Please post the output of
    lsusb -d 1532: -v
```

```
(I'll add when I reboot into Linux)
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
[image]

iKarith, 2020-05-09:
Just to add that the mouse dock is in fact the same one that comes with the Viper Ultimate and presumably other wireless mice, 1532:007e, and that the human-readable name is neither NUL-terminated nor spaced out to the full length.

I note above,

```
Bus 001 Device 010: ID 1532:007e Razer USA, Ltd Razer Goliathus Extended Chroma
```

Mine shows:

```
Bus 002 Device 015: ID 1532:007e Razer USA, Ltd USB2.1 Hub
```

The name changes based upon device connection order. Bad Razer. 😉

(I'm not conveniently able to do a pcap for the moment, just noting that a larger group of people likely can.)

Arilas, 2020-05-25:
Support for this is done in #1152

lah7, 2021-10-31:
This was implemented in #1321

closed 2021-10-31 by lah7