openrazer/openrazer #1020: Key spamming on blackwidow elite keyboard

ddickstein, 2020-02-02:
I'm not sure this is an openrazer issue or a product issue (in which case I might just need to get a new keyboard). On my blackwidow elite keyboard sometimes I'll quickly press and release a key and the release won't be registered, causing the key to spam (like aaaaaaaaaaaaaaaaaaaaaaaaaaaa). I normally operate with a fast key repeat rate (xset r rate 200 50) but because I've been running into problems I've tried slowing the trigger time down to 500. I'm still encountering issues and 500 is already frustrating, so I don't want to try going lower (and it's honestly hard to say if raising this number is helping since the issue is sporadic).

Running openrazer-daemon 2.7.0 on 5.4.15-arch1-1.

I don't recall having this issue with the same keyboard connected to my laptop (macbook pro) which was also using a slightly earlier version of arch. Right now I'm running on a custom build, and can provide component specs if relevant.

ddickstein, 2020-02-02:
I can also confirm that it is always the lowercase version of the key that is spammed, so releasing shift is being recognized. For example, if I press and release G I might get gggggggggggggg. If I press and release % I might get 5555555555555.

z3ntu, 2020-02-08:
Have you tried uninstalling openrazer and checking if the issue is still present?

ddickstein, 2020-02-27:
You're right - the issue persists after removing openrazer. I didn't encounter this previously though when using the same keyboard when attached to my laptop (also running arch). Where would you recommend I start investigating the problem?

z3ntu, 2020-03-14:
No idea, sorry :/

closed 2020-03-20 by ddickstein

IvanVojtko, 2020-06-15:
I know it's closed already, but I had this issue on a different keyboard. It can be issue with polling. Blackwidow elite uses 1000hz and I had the same issue with this polling rate. Try to decrease polling rate and see if it helps.