openrazer/openrazer #1022: Cant detect my device(User is not a member of the plugdev group)

1m-N00b, 2020-02-05:
# Problem
When I started polychromatic, I got error(**someting went wrong**) and started the **restarted daemon**.Then I got the error(**Deamon cant be lelaunched**), so I check the log(~/.local/share/openrazer/logs/razer.log) and run **openrazer-demon -Fv**.
Here is the result.
**logs**
`2020-02-06 00:30:00 | razer                          | CRITICAL | User is not a member of the plugdev group`
**deamon**
` openrazer-daemon  -Fv                                                 
Starting daemon.
2020-02-06 01:29:59 | razer                          | CRITICAL | User is not a member of the plugdev group
Stopping daemon.`

#Done 
i search this problem.then i found same problem(https://github.com/openrazer/openrazer/issues/761)
So i check linux module `cd /usr/lib/modules/ `
result is only one(**5.5.1-arch1-1/**).

#distribution
Arch linux  5.4.10-arch1-1
i checked my device`lsusb | grep '1532:'`
Here is the result.
`Bus 001 Device 089: ID 1532:0216 Razer USA, Ltd BlackWidow X Chroma
Bus 001 Device 002: ID 1532:005c Razer USA, Ltd Razer DeathAdder Elite`

damianatorrpm, 2020-02-05:
After adding your user to plugdev group you need to log-out and log-in again.

1m-N00b, 2020-02-05:
oops It was a simple problem.
thx

closed 2020-02-08 by z3ntu