openrazer/openrazer #1025: Blade Late 2017 Set Profile at Startup?

ign1s-fatuus, 2020-02-09:
I have been using openrazer for a while now but not without some annoyances. It seems that after every reboot I have to reset the keyboard profile by hand. Is there any way of making profiles persist between reboot? Or, if that is not a possibility, is there any way of setting the profile automatically at boot? I see that there looks to be some interesting python examples of srcripts but I have been unable to find any API documentation detailing functions that might help me script my way out of this endeavor. 

Any help would certainly be appreciated.

lah7, 2020-02-09:
By "profile", do you mean a hardware effect (static, wave, ripple) or a _custom effect_ where you've set colours for different keys in a front-end?

When #888 is merged, OpenRazer will retain the last effect across reboots.

Front-ends (Polychromatic, RazerGenie, razerCommander) currently let you create colour mappings for keys but they currently don't have their own methods of setting one automatically. Older Razer firmware were able to retain their LED states across reboots.

At the moment, there isn't much documentation (there was a [feature_docs](https://github.com/openrazer/openrazer/tree/feature_docs) branch) but the Python code is relatively simple:

```python
#!/usr/bin/python3
import openrazer.client
device = openrazer.client.DeviceManager().devices[0]

print(device.fx.advanced.rows)
print(device.fx.advanced.cols)

device.fx.advanced.matrix[0,0] = (0, 255, 0)   # Red
device.fx.advanced.matrix[1,0] = (255, 0, 0)   # Green
device.fx.advanced.matrix[2,0] = (0, 0, 255)   # Blue
device.fx.advanced.draw()
```

You can tweak this, save it to a `.py` file and add it to your desktop environment's auto start applications.

ign1s-fatuus, 2020-02-10:
I will have to look at the python code and play around with it. 

By profile I am referring to the application profile used/generated via polychromatic. I generally like to set colors for various vim keys (hjkl, &c.). Here is the contents of the json file from which the profile is loaded:

`{"name": "xmonad_kdb", "icon": "/usr/share/polychromatic//img/profile-default.svg", "rows": {"0": [[0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255]], "1": [[0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255]], "2": [[0, 255, 255], [0, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [0, 255, 255], [255, 0, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255]], "3": [[0, 255, 255], [255, 0, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 0, 255], [255, 0, 255], [255, 0, 255], [255, 0, 255], [255, 255, 255], [255, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255]], "4": [[0, 255, 255], [0, 255, 255], [0, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [255, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255]], "5": [[0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255], [0, 255, 255]]}}% `

Any way, thank you and I will keep in plugging away at this issue for a bit longer. ... then on to the fn key problem which likewise plagues me

lah7, 2020-02-10:
No problem! I figured you may be using Polychromatic as it's the only front-end that calls them "application profiles" (for now). Later this year, its v1.0.0 release aims to improve effects/profiles functionality.

---

> then on to the fn key problem which likewise plagues me

If it's the issue where pressing the FN keys lights up the F1-F12 keys but doesn't turn them off afterwards, that's down to the firmware, and it doesn't know its previous state. It needs "something" to send the frames again, which is what Synapse would normally do as it controls everything in software.

A workaround is to add to this to your Python script -- so every second, custom frames are sent:

```python
import time
while True:
    [your existing code intended 4 spaces]
    time.sleep(1)
```

ign1s-fatuus, 2020-02-15:
You got the Fn key issue right on the nose. I was thinking of doing something similar as per the code snippet offered above. I am trying to figure out a way to achieve the same thing but done a little more efficiently. One thought might be to send a custom frame every time the fn+fx key is triggered. As I generally use xmonad as my window manager I am thinking of mapping XF86[keys] to likewise trigger a reset and thus avoid a while loop that 99.9% of the time simply sets the keyboard layout to what it already is. 

This will not work perfectly though as xev results produce that fn+f4=p and fn+10/11 do not produce a key result (are hardwired to adjust kbd brightness).

I guess I am of two minds. For one I'd like an elegant solution that does as little as possible to use up resources. On the other hand, how much of my resources are really being used by a while loop that is sending a custom frame once a second?

Again, thanks for the help and for pointing me in the right direction.

closed 2020-07-18 by stale[bot]