openrazer/openrazer #1027: Support for Razer_Razer_Cynosa_Lite

AnlorNode, 2020-02-10:
### Device Info

* Name: Vladimir
* Product Number: rz03-02741500-r3r1
* Model Number: RZ03-0274

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x023f
  bcdDevice            2.00
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval
```
Please post the output of
    ls -lah /dev/input/by-id/
```
drwxr-xr-x 2 root root 220 лют  9 22:57 .
drwxr-xr-x 4 root root 660 лют 10 09:39 ..
lrwxrwxrwx 1 root root  10 лют  9 22:31 usb-HORI_CO._LTD_DEGE_101_Optical_Gaming_Mouse-event-if01 -> ../event14
lrwxrwxrwx 1 root root  10 лют  9 22:31 usb-HORI_CO._LTD_DEGE_101_Optical_Gaming_Mouse-event-mouse -> ../event12
lrwxrwxrwx 1 root root  10 лют  9 22:31 usb-HORI_CO._LTD_DEGE_101_Optical_Gaming_Mouse-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root   9 лют  9 22:31 usb-HORI_CO._LTD_DEGE_101_Optical_Gaming_Mouse-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 лют  9 22:57 usb-Razer_Razer_Cynosa_Lite-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 лют  9 22:57 usb-Razer_Razer_Cynosa_Lite-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 лют  9 22:57 usb-Razer_Razer_Cynosa_Lite-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 лют  9 22:57 usb-Razer_Razer_Cynosa_Lite-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 лют  9 22:57 usb-Razer_Razer_Cynosa_Lite-if02-mouse -> ../mouse0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
###########
https://hard.rozetka.com.ua/razer_rz03_02741500_r3r1/p179599197/?gclid=Cj0KCQiAm4TyBRDgARIsAOU75sog1ohyWZhV6ooWEjD8MEcdzYQ5hD3N_4yFj9G_zbGL7yZ0rxYndQwaAu1SEALw_wcB

AnlorNode, 2020-02-10:
[image]

oviradoi, 2020-04-04:
I have captured a pcapng file for this device and I have recorded a video of me changing the settings in Razer Synapse. 

You can use the timestamp in the pcapng file and the timestamp in the video in order to correlate the  settings in Synapse and the packets being captured.

I am attaching the capture and the video here. I hope this helps you add support for this keyboard.

[win7.wemb.zip](https://github.com/openrazer/openrazer/files/4431494/win7.wemb.zip)
[capture.pcapng.zip](https://github.com/openrazer/openrazer/files/4431496/capture.pcapng.zip)

commit 6c71ccfc (2020-07-18): Add support for Razer Cynosa Lite

Fixes #1027
closed 2020-07-18 by z3ntu