openrazer/openrazer #1030: Installation on Ubuntu 20.04 not possible.

pallebone, 2020-02-12:
### Distribution

### Support

Installation causes apt dependencies on latest Ubuntu/varients LTS release (to be released in April this year).

Output:

aragorn@aragorn:~$ sudo apt install openrazer-meta
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help resolve the situation:

The following packages have unmet dependencies:
 libgcc-s1 : Breaks: cryptsetup-initramfs (< 2:2.2.2-3~) but 2:2.2.2-2ubuntu1 is to be installed
 libgcc-s1:i386 : Breaks: cryptsetup-initramfs (< 2:2.2.2-3~) but 2:2.2.2-2ubuntu1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

pallebone, 2020-02-12:
Resolution - manually install cryptsetup-initramfs:
The following packages have unmet dependencies:
 libgcc-s1 : Breaks: cryptsetup-initramfs (< 2:2.2.2-3~) but 2:2.2.2-2ubuntu1 is to be installed
 libgcc-s1:i386 : Breaks: cryptsetup-initramfs (< 2:2.2.2-3~) but 2:2.2.2-2ubuntu1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
aragorn@aragorn:~$ ^C
aragorn@aragorn:~$ sudo apt-get in
indextargets  install       
aragorn@aragorn:~$ sudo apt-get install cryptsetup-initramfs 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  cryptsetup-run
The following packages will be upgraded:
  cryptsetup-initramfs cryptsetup-run
2 upgraded, 0 newly installed, 0 to remove and 79 not upgraded.
Need to get 28.2 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 cryptsetup-run all 2:2.2.2-3ubuntu1 [5,340 B]
Get:2 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 cryptsetup-initramfs all 2:2.2.2-3ubuntu1 [22.9 kB]
Fetched 28.2 kB in 0s (149 kB/s)                
Preconfiguring packages ...
(Reading database ... 254524 files and directories currently installed.)
Preparing to unpack .../cryptsetup-run_2%3a2.2.2-3ubuntu1_all.deb ...
Unpacking cryptsetup-run (2:2.2.2-3ubuntu1) over (2:2.2.2-2ubuntu1) ...
Preparing to unpack .../cryptsetup-initramfs_2%3a2.2.2-3ubuntu1_all.deb ...
Unpacking cryptsetup-initramfs (2:2.2.2-3ubuntu1) over (2:2.2.2-2ubuntu1) ...
Setting up cryptsetup-run (2:2.2.2-3ubuntu1) ...
Setting up cryptsetup-initramfs (2:2.2.2-3ubuntu1) ...
update-initramfs: deferring update (trigger activated)
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.133ubuntu14) ...
update-initramfs: Generating /boot/initrd.img-5.4.0-12-generic

aragorn@aragorn:~$ sudo apt install openrazer-meta
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  gcc-10-base gcc-10-base:i386 libatomic1 libatomic1:i386 libblas3 libcc1-0 libgcc-s1 libgcc-s1:i386 libgcc1 libgfortran5 libgomp1
  libitm1 liblapack3 liblsan0 libquadmath0 libstdc++6 libstdc++6:i386 libtsan0 libubsan1 openrazer-daemon openrazer-driver-dkms
  python3-daemonize python3-notify2 python3-numpy python3-openrazer python3-pyudev xautomation
Suggested packages:
  openrazer-doc python-daemonize-doc gfortran python-numpy-doc python3-dev python3-pytest python3-numpy-dbg
The following NEW packages will be installed:
  libblas3 libgfortran5 liblapack3 openrazer-daemon openrazer-driver-dkms openrazer-meta python3-daemonize python3-notify2
  python3-numpy python3-openrazer python3-pyudev xautomation
The following packages will be upgraded:
  gcc-10-base gcc-10-base:i386 libatomic1 libatomic1:i386 libcc1-0 libgcc-s1 libgcc-s1:i386 libgcc1 libgomp1 libitm1 liblsan0
  libquadmath0 libstdc++6 libstdc++6:i386 libtsan0 libubsan1
16 upgraded, 12 newly installed, 0 to remove and 63 not upgraded.
Need to get 8,822 kB of archives.
After this operation, 33.2 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ca.archive.ubuntu.com/ubuntu focal/main i386 gcc-10-base i386 10-20200211-1ubuntu1 [18.8 kB]
Get:2 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 gcc-10-base amd64 10-20200211-1ubuntu1 [18.8 kB]
Get:3 http://ca.archive.ubuntu.com/ubuntu focal/main i386 libstdc++6 i386 10-20200211-1ubuntu1 [542 kB]
Get:4 http://ppa.launchpad.net/openrazer/stable/ubuntu focal/main amd64 openrazer-driver-dkms all 2.7.0~ubuntu20.04.1 [48.7 kB]
Get:5 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 libstdc++6 amd64 10-20200211-1ubuntu1 [502 kB]
Get:6 http://ca.archive.ubuntu.com/ubuntu focal/main i386 libatomic1 i386 10-20200211-1ubuntu1 [9,884 B]
Get:7 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 libatomic1 amd64 10-20200211-1ubuntu1 [9,308 B]
Get:8 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 libubsan1 amd64 10-20200211-1ubuntu1 [137 kB]
Get:9 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 libtsan0 amd64 10-20200211-1ubuntu1 [320 kB]
Get:10 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 libquadmath0 amd64 10-20200211-1ubuntu1 [147 kB]
Get:11 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 liblsan0 amd64 10-20200211-1ubuntu1 [145 kB]
Get:12 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 libitm1 amd64 10-20200211-1ubuntu1 [26.3 kB]
Get:13 http://ca.archive.ubuntu.com/ubuntu focal/main amd64 libgomp1 amd64 10-20200211-1ubuntu1 [101 kB]
Get:14 http://ca.archive.ubuntu.com/ubuntu focal/main i386 libgcc-s1 i386 10-20200211-1ubuntu1 [49.8 kB]
[... 153 lines cut ...]
   - Installing to /lib/modules/5.4.0-12-generic/updates/dkms/

depmod......

DKMS: install completed.

ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'
Setting up libstdc++6:amd64 (10-20200211-1ubuntu1) ...
Setting up libstdc++6:i386 (10-20200211-1ubuntu1) ...
Setting up liblsan0:amd64 (10-20200211-1ubuntu1) ...
Setting up libitm1:amd64 (10-20200211-1ubuntu1) ...
Setting up libtsan0:amd64 (10-20200211-1ubuntu1) ...
Setting up python3-daemonize (2.4.7-4) ...
Setting up liblapack3:amd64 (3.9.0-1) ...
update-alternatives: using /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3 to provide /usr/lib/x86_64-linux-gnu/liblapack.so.3 (libl
apack.so.3-x86_64-linux-gnu) in auto mode
Setting up openrazer-daemon (2.7.0~ubuntu20.04.1) ...
Setting up python3-numpy (1:1.17.4-5ubuntu2) ...
Setting up libubsan1:amd64 (10-20200211-1ubuntu1) ...
Setting up libcc1-0:amd64 (10-20200211-1ubuntu1) ...
Setting up python3-openrazer (2.7.0~ubuntu20.04.1) ...
Setting up openrazer-meta (2.7.0~ubuntu20.04.1) ...
Processing triggers for man-db (2.9.0-2) ...
Processing triggers for libc-bin (2.30-0ubuntu3) ...
aragorn@aragorn:~$

closed 2020-02-12 by pallebone

SourceSkyBoxer, 2020-04-25:
Hello me same problem with f**king Mono Runtime 6.8.0 ( it is wrong version )