openrazer/openrazer #1031: Macro keys not working for BlackWidow Ultimate 2013

RiverMatt, 2020-02-16:
### Distribution
Linux Mint 19.3

### Support

- [Y] Have you tried rebooting?
- [Y] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The macro keys are not working on my BlackWidow Ultimate 2013. I've installed the driver, confirmed that it works, and /sys/bus/hid/drivers/razerkbd/0003:1532:011A.000F/device_mode shows 3:0 (which should be driver mode, right?). I've tried RazerGenie and Polychromatic, which let me adjust the lighting (this does work), but I cannot record macros with those programs as far as I can tell.

As far as I can tell, the only way to record macros is via the on-the-fly method. Pressing Fn+F9 causes the flashing M to appear, I press a macro key (M1-M5), the light goes solid, then I type a macro, finishing by pressing Fn+F9. Macro light goes away. However, when I press the M-button I recorded it on, nothing happens.

Getting these macro keys to work could be quite handy for vim...

Hopefully I'm not missing something silly, but this appears to be a problem. Thank you.

OfficialStamper, 2020-06-23:
Bump

Linux Mint 20

Razer Blackwidow Ultimate 2012

I never get the Flashing (M) with either FN+(M) nor FN+F9

Unable to record/use macros at all

closed 2021-06-22 by github-actions[bot]

z3ntu, 2026-02-15:
Please don't necro-bump issues. And macros are essentially unsupported and will be removed soon, see #2337. Use https://github.com/sezanzeb/input-remapper