openrazer/openrazer #1037: None of my Devices Pop Up

LakshGup5, 2020-02-27:
When I try to open the app Razer Genie, it shows an error screen. However, I have a full razer setup. I am running a custom build PC with Linux Mint on it. Also, can you add support for the Lian Li razer case?

lah7, 2020-02-27:
OpenRazer doesn't support Lian Li Razer Case. Do you know if this supports Razer Synapse?

If it does, please create a new issue and follow the "Device Support" template. It is necessary to capture pcaps under Windows (even if it's just a VM) so it's known what protocol it uses.

Do you have other devices connected that are not appearing in the list?

LakshGup5, 2020-02-29:
The Lian Li is supported in Synapse. Also, My main problem was the things before the case. Why isn't it detecting it? And yes. It is.

lah7, 2020-03-01:
Devices not appearing in the daemon are usually because:

* The Linux headers are not installed.
* and/or your user account isn't added to the `plugdev` group, 
* and/or secure boot is enabled and the driver refuses to load. 

What distro are you using?

Take a look at the [Troubleshooting page (Kernel modules and Daemon sections)](https://github.com/openrazer/openrazer/wiki/Troubleshooting#kernel-modules-loading) - what's the output of these commands?

closed 2020-08-02 by stale[bot]