openrazer/openrazer #1038: Support for macOS

0x77dev, 2020-03-02:
### Distribution

### Support

- [ NO NEEDED IN THIS CASE ] Have you tried rebooting?
- [ YES ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

redaxmedia, 2020-03-12:
I suggest to start with extending the `.travis.yml`file and see what happens:

```
os:
 - linux
 - osx
```

lah7, 2020-03-12:
Duplicate of #959 (and #991). Please see #959 for details.

Simply put, it's not possible for OpenRazer in its current form to run on Windows/Mac because it depends on a Linux-only driver. There's @z3ntu's [razer_test](https://github.com/z3ntu/razer_test/) which handles everything in user space, eliminating the need for a driver. It's a work-in-progress, and is already making progress to run on Windows/Mac. Eventually it'll supersede OpenRazer 2.x.

closed 2020-03-12 by lah7